OctoPrint/OctoPrint #2957: [Request] New plugin mixin: PrinterPlugin

tjjfvi, 2018-12-15:
**Is your feature request related to a problem? Please describe.**
When developing my first plugin, which was to count `echo:busy` responses from the printer and notify if the number passed a threshold, it was unclear how to receive printer responses. I had to go to the "Internal Modules" section of the docs to see the class `PrinterCallback`.

**Describe the solution you'd like**
It would be nice if there was a simple mixin that facilitates interacting with the printer. It could simply be (I think):
```python
class PrinterPlugin(octoprint.plugin.core.Plugin, octoprint.printer.PrinterCallback):
    def __init__(self):
        self._printer.register_callback(self)
```

**Describe alternatives you've considered**
One could also just add a section in the plugin docs directing people to the `PrinterCallback` docs.

**Additional context**
Add any other context or screenshots about the feature request here.

foosel, 2019-03-12:
That's actually what the [`octoprint.comm.protocol.gcode.<phase>` family of hooks](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase) and [`octoprint.comm.protocol.gcode.received`](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-received) are for (well, that and rewriting input/output).

I'm not sure if adding a mixin that combines those would not complicate things rather than simplifying them.

I'd be happy about suggestions on how to make them easier discoverable. In general I'd be happy about help with more plugin tutorials and/or "cookbook" like guides on how to achieve certain common tasks.

Any suggestions?

tjjfvi, 2019-03-13:
Personally, I liked the `PrinterInterface` and `PrinterCallback` apis; those hooks seem to be much less abstracted (though, for my use case, seems they would have done the job I needed); are the "Internal Modules" included in the public api for `semver` purposes, or could they change on a minor/patch update?

The cookbook sounds like a great idea, and, as the mixin was intended mainly to make it more visible/slightly less work, sounds like a better idea.
Do you think you would put it in the docs or in the wiki or somewhere else?

foosel, 2019-04-05:
> The cookbook sounds like a great idea, and, as the mixin was intended mainly to make it more visible/slightly less work, sounds like a better idea.
> Do you think you would put it in the docs or in the wiki or somewhere else?

Good question. Docs probably, and then heavily linked to.

Also sorry, I somehow missed the notification about this.