OctoPrint/OctoPrint #2958: Critical error in 1.13.10 introducing filament change request (pauses) - without being requested!

srehne, 2018-12-16:
I am running raspberry pi 3B+ with 1.13.10 and OctoPi version 0.15.1
My printer is a Prusa i3 MK3 - with latest software.

I suddenly am getting pauses in my print - requesting change of filament.

Note that I have - so far - never used multi filement commands in any way.
Most of my different prints now how begun
[4xboks_holder_gcode.txt](https://github.com/foosel/OctoPrint/files/2683685/4xboks_holder_gcode.txt)

 to stop in the middle of the print and ask for a change of filement.

I am using Slic3r Prusa Edition - 1.41.2+ to generate the gcode and in the past I have not had any of this kind of problem before.

I include my gcode file here - if you can debug it perhaps to see that no change of filement is included.
I've searched the file for M603 M6xx M0 just to try and find a problem in the gcode file. 

_**Next problem:**_
I can't find any standard way to downgrade to version 1.13.9. Anyone who has an "official" recipe?
I've seen some "git checkout" commands - but they describe a manual start of the server - I just want my OctoPrint to go permanently back to 1.13.9.

[4xboks_holder_gcode.txt](https://github.com/foosel/OctoPrint/files/2683686/4xboks_holder_gcode.txt)

foosel, 2018-12-17:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * whether the issue persists in safe mode - if not it's one of your third party plugins
  * your [exact version of OctoPrint](https://discourse.octoprint.org/t/how-can-i-find-out-the-version-of-octoprint-or-octopi-i-am-running/204/1)
  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * contents of the terminal tab, or even better a [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * the contents of [your JS error console](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers)

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

Instructions on how to downgrade can be found [in the FAQ](https://discourse.octoprint.org/t/how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image/205). Since 1.3.10 didn't introduce any kind of "sudden filament change" (or any kind of "let's just send random commands in the middle of a print that might nuke it" functionality) I fear that won't solve the problem since it's most likely caused by either a third party plugin or a configuration on your end. Try safe mode & provide all the requested logs.

srehne, 2018-12-17:
Hi again.

I've done a safe mode print of the file and it worked!
So my problem is within one of my plugins.
I believe I only have "standard" plugins - if not please let me know.

Here is my list of plugins:
Page 1:
[image]

Page 2:
[image]

Page 3:
[image]

I believe that I can solve my problem by just disabling all your plugins.

My version of OctoPrint - taken from the website - says:
OctoPrint 1.3.10 running on OctoPi 0.15.1

All my octoprint.log files are here:
[octoprint.zip](https://github.com/foosel/OctoPrint/files/2685785/octoprint.zip)

All octoprint serial.log files are here:

[octoprintserial.zip](https://github.com/foosel/OctoPrint/files/2685789/octoprintserial.zip)

Here is my contents of my JS console:
[image]

Do you have an idea of which of your plugins might be the "best candidate for failure" compared to the filament change problem? I'd be happy to only disable one and print again.
But since the print is approx 7 hours - I'd not like to do this test too often.
Remember I have to do the filement change procedure.
And if I print during the night - the bed will keep the 90degree temperature and "burn energy on nothing". ;-)

foosel, 2018-12-17:
All of these plugins are also enabled in safe mode since they are bundled (recognizable by the little supposed-to-be-a-package symbol). Only third party plugins (not bundled with OctoPrint) will be disabled in safe mode. None of the bundled plugins send filament swap commands to the printer (in fact, unless I'm misremembering none of them in fact send *any* commands to the printer ever). 

It could also be your configured GCODE scripts, a third party client (in case you use anything but the default web interface), etc. Or it could just be something in your firmware deciding to trigger a filament swap. Impossible to say without a `serial.log`, which you sadly hadn't enabled. Please enable it, reproduce the problem, provide the resulting file.

Does the weird filament swap stuff only happen on this 7h long print? If not I suggest to reproduce it with something shorter to debug this. Since you have a MK3, try also to disable its filament probe - it could be misfiring here for some reason, causing this issue. Also state your firmware version please, just in case.

I also have noticed that your instance is listening on a public IP (and gets probed by malicious players for random webserver weaknesses). [**Do not expose your OctoPrint instance publicly on the internet!**](https://octoprint.org/blog/2018/09/03/safe-remote-access/). Rectify this situation immediately!

slackerbob, 2018-12-17:
I can confirm this also happening to me.

2+ hours into a print, octoprint pauses, nozzle homes itself (X, Y and Z, damaging prints in the process), and it shows no error.

Hitting resume from this point just damages the print further, due to the print head being in the wrong location.

My version of octoprint is the same:
OctoPrint 1.3.10 running on OctoPi 0.15.1

[octoprint.log](https://github.com/foosel/OctoPrint/files/2687782/octoprint.log)

The most recent pause was at:
`2018-12-17 10:34:23,020 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
`

I do have the octoprint.plugins.excluderegion plugin running (very useful!).

If you need a serial.log from me, I'll have to create one. I'd like to avoid that, as that would mean wasting 2 hours worth of filament again, but if it's necessary, let me know and I'll do it.

srehne, 2018-12-18:
Hi again

I can now confirm that the problem also happens in Safe mode - with the access from "the public" turned off. (open inside my wifi - but no access from outside - firewall blocked).
Yesterday the same print - that previously had 2 stops ... had 3 stops.
I will read your guide more carefull regarding the serial.log - which I thought by sending you files was enabled. AND look closer at firmware etc. and get back to you later when I run the file again.

foosel, 2018-12-18:
> If you need a serial.log from me, I'll have to create one. I'd like to avoid that, as that would mean wasting 2 hours worth of filament again, but if it's necessary, let me know and I'll do it.

It is necessary I fear. Without seeing what is going to and coming from the printer it is pretty impossible to debug issues like this.

slackerbob, 2018-12-18:
Printed in safe mode, with the serial log turned on:

OctoPrint 1.3.10 running on OctoPi 0.15.1

[octoprint.log](https://github.com/foosel/OctoPrint/files/2692164/octoprint.log)

[serial.log](https://github.com/foosel/OctoPrint/files/2692165/serial.log)

[PLA-FILAMENT_TUBE-1h.gcode.txt](https://github.com/foosel/OctoPrint/files/2692188/PLA-FILAMENT_TUBE-1h.gcode.txt)

Same issue...also, it seems that all these fails are at roughly the same height, at least in my case.

I do have a filament sensor, but this problem has only started since upgrading octoprint to 1.3.10.

FanDjango, 2018-12-19:
I found
```
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 170
M83
G1 F1500 E-5
G1 F300 Z43.45
G1 F9000 X20 Y20
M104 S0; standby temperature
M0;Do the actual pause
```
in your GCODE.
The location of that fragment corresponds to the place in the serial.log where your printing stops and comms with the printer are lost.

What is: ***PauseAtHeight.py***?

slackerbob, 2018-12-19:
You both are quite right!

I apologize for being an idiot!

I looked in Cura, and somehow that extension was turned on (In both my Cura installs, somehow).

Thank you very much for all your help, and once again, I apologize for blaming Octoprint for this! (While trying to diagnose the problem, I found this thread, having the same problem as me, and just assumed that it was the same cause.)

FanDjango, 2018-12-22:
@srehne:
Still no serial.log to look at, but I have been poring over your octoprint log. I have found numerous occurences of the following:
```
| Last lines in terminal:
| Recv: ok
| Send: N306155 G1 X54.977 Y83.105 E0.01991*81
| Recv: ok
| Send: N306156 G1 X55.598 Y82.484 E0.03001*81
| Recv: ok
| Send: N306157 G1 X55.598 Y81.903 E0.01990*82
| Recv: ok
| Send: N306158 G1 X54.977 Y82.523 E0.03001*95
| Recv: ok
| Recv: fsensor_stop_and_save_print
| Send: N306159 G1 X54.977 Y81.941 E0.01991*87
| Recv: fsensor_oq_meassure_start
| Recv: echo:Enqueing to the front: "G1 E-3 F200"
| Recv: echo:Enqueing to the front: "G1 E3 F200"
| Recv: fsensor_oq_meassure_stop, 5 samples
| Recv:  st_sum=750 yd_sum=26 er_sum=2 er_max=1
| Recv:  yd_min=31 yd_max=7 yd_avg=6 sh_avg=1
| Recv: fsensor_err_cnt = 0
| Recv: fsensor_restore_print_and_continue
| Recv: Resend: 306144
```
and also
```
| Last lines in terminal:
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: paused for user
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: echo:enqueing "M220 S100"
| Recv: fsensor_restore_print_and_continue
```
It seems there was/is? an issue with the filament sensor on Prusa-Firmware 3.4.0. I know you have Prusa-Firmware 3.4.1, but in the posts on the internet I have seen no resolution or anything. Here is the relevant link: [https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/firmware-3-4-0-fsensor-message-and-printer-pauses-t24713.html](https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/firmware-3-4-0-fsensor-message-and-printer-pauses-t24713.html). As you can see there, someone complains about the same symptoms.

***Perhaps you could disable your filament sensor as a temporary workaround to see if this is relevant?***

Still would be interesting to see serial.log, though.

srehne, 2018-12-26:
Prusa released 3.5.1 on Dec. 19. with these changes: 

**CHANGES IN 3.5.1**
Patch release
Button click fix
M220 speed factor
Z-levelling fix
Pause print -> stop print fix

I think my problem might have been addressed by Prusa! 
I will continue testing and see if this update has fixed my problem.

foosel, 2019-01-10:
So... has it?

srehne, 2019-01-21:
Hi again
Sorry for the delay. I finally got to print again - and had a big disaster regarding filament on the print nozzle. Finally got it cleaned - and now I have made 2 prints - without the problem!

So I believe the problem is over since the 3.5.1 update!! 

Sorry for addressing the issue in this forum. I didn't think the Prusa firmware could be the reason.

closed 2019-01-21 by srehne