OctoPrint/OctoPrint #2959: Release Announcement Sticky note

tonymaglioni, 2018-12-16:
#### What were you doing?

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

tonymaglioni, 2018-12-16:
[image]

foosel, 2018-12-17:
That isn't a new feature introduced with 1.3.10, for some reason (network outage?) your OctoPrint instance wasn't able to fetch announcements for a long while.

Your screen resolution of 1024x768 is as bit on the small side these days, and I simply didn't anticipate that here - the action buttons for the notication are at its bottom. I'll look into fixing that in 1.3.11. Until then I suggest to access OctoPrint from a PC with a higher resolution and close the notification there.

foosel, 2019-01-15:
Solved by the above commit that adds a `max-height` to all notifications that is calculated from the available viewport size:

[image]

Available on `maintenance`, soon `devel`, to be released with 1.3.11.

commit 918b7018 (2019-01-15): Set max-height on PNotify notifications

Fixes #2959
closed 2019-05-14 by foosel