OctoPrint/OctoPrint #2961: [Request]Activate python script via button

jakobmaraton, 2018-12-17:
I have been searching for a while, to be able to add a button to activate a script. Preferable this would make it possible to activate gpio pins on the raspberry pi. The already existing addons have what I know very limited functionality. The end goal is to control a servo reliably from the GPIO, so if there is any other suggestion I am open. I have tested "Pan/Tilt" and "enclosure plugin" with no success.

jakobmaraton, 2018-12-20:
Ok, please correct me if I am wrong. To activate a python script from OctoPrint, I first need to convert the file to an .exe, and then via "system command editor" specify where the file is located.

foosel, 2019-01-10:
> To activate a python script from OctoPrint, I first need to convert the file to an .exe, and then via "system command editor" specify where the file is located.

That depends entirely on your OS and you haven't made clear what you are even trying to do. Maybe you should rather open a topic on the [community forum](https://discourse.octoprint.org) and describe what you are trying to achieve and what you already tried, that way it might be possible to show you how to solve whatever is the problem you are trying to solve with what is already implemented.

closed 2019-01-10 by foosel