OctoPrint/OctoPrint #2962: [Request] Log file to easy deletable!

dahuafschmied, 2018-12-19:
Hi Gina!

I think I dicovered have a little bug and I want to give you a perfect bug report. BUT! My mouse aiming was a little disturbed by my cat and i hit the delete button instead of the download button. They are real close together. So the log is gone and I have to wait some days to reproduce the random bug.

My suggestion:
Please add a "Are you sure?" (Yes / No / HabAngst / Call Telefonseelsorge) button to the logs dialog.

Merry Christmas and a happy new year
dahuafschmied

foosel, 2019-01-10:
I like the two additional answer options ;)

foosel, 2019-01-15:
Implemented by the above commit, already on `maintenance`, soon `devel`, to be part of 1.3.11.

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel