OctoPrint/OctoPrint #2963: can't select option if changed language display

fred2088, 2018-12-21:
select another language, display without any problem after re-login,

but can't change options page from any tab of octoprint menu !!!

Thanks, Regards

fred2088, 2019-01-06:
sloved !!! the translate replace the "<" ">" charaters into "＜", "＞", which destruct the <code> and </code> pairs ...

closed 2019-01-06 by GitIssueBot

chenxiaowei107, 2020-04-28:
> sloved !!! the translate replace the "<" ">" charaters into "＜", "＞", which destruct the ` and ` pairs ...

Hello ,I think i got the same problem , please tell me more information,Thank you very much.

fred2088, 2020-04-29:
for zh_CN/zh_HK/zh_TW, you can download from https://www.transifex.com/octoprint/octoprint/

chenxiaowei107, 2020-04-29:
> for zh_CN/zh_HK/zh_TW, you can download from https://www.transifex.com/octoprint/octoprint/

Yes I download It and build in the OctoPi ,It can worked in 1.3.0,But if i update to 1.4.0 ,the menu is not working well . The Setting Menu Typesetting is not working well.

fred2088, 2020-04-29:
which one you download CN/HK/TW ?

chenxiaowei107, 2020-04-30:
> which one you download CN/HK/TW ?

zh_CN,
I download the newest Po file in transifex
also I use the command “python setup.py babel_new --locale=zh_CN“ creat a zh_CN po file 

I find There are some difference for two file

fred2088, 2020-04-30:
my work is zh_HK which using TC and with HK word pharse ...

chenxiaowei107, 2020-04-30:
> my work is zh_HK which using TC and with HK word pharse ...

I will use zh_HK to try.

chenxiaowei107, 2020-04-30:
[image]
if i use the zh_CN the menu will like this