OctoPrint/OctoPrint #2965: [Request] Custom Event Support

tjjfvi, 2018-12-22:
**Is your feature request related to a problem? Please describe.**
It is not related to a problem.

**Describe the solution you'd like**
If OctoPrint could support custom events, triggered by plugins, that would be great. If it already (documented or not) supports this, that's even better; I don't need to wait :)

**Describe alternatives you've considered**
I have not thought of any alternatives.

**Additional context**
I recently created a plugin for OctoPrint that notififies via IFTTT when the printer sends N or more busy signals. Obviously, that is a very specialized plugin; it would most likely be much more useful to have two plugins: one that adds a EchoBusy event and one that sends some or all events to IFTTT.

I have looked into the source/docs some and found that a plugin could run `self._event_bus.fire("MyHappyCustomEvent", dict(payload_key=payload_val))` and it would (seemingly) work and not destroy OctoPrint, but I haven't thoroughly tested it.

foosel, 2019-01-10:
In theory it should work to set the new event on `octoprint.events.Events`, that will make it register in the client too, but I agree, a more straightforward hook or something to accomplish that is really needed.

tjjfvi, 2019-01-10:
Ah, I hadn't seen that; maybe just throwing a paragraph about using that and the `self._event_bus.fire` to make custom events is enough?

foosel, 2019-01-15:
> maybe just throwing a paragraph about using that and the `self._event_bus.fire` to make custom events is enough?

In theory, but monkey patching is a bit hackish. By providing a defined hook namespace collisions could also be avoided (imagine two plugins deciding they want to define an event `SOMETHING_HAPPENED`...).

tjjfvi, 2019-02-25:
@foosel: As I read it, it seems that you can only pregister events for `plugin_myhappyplugin_my_happy_event_name`. Would it be possible to also allow `my_happy_event_name` as an alias (only to be used by user-facing apis), assuming no conflicts?

foosel, 2019-03-25:
> Would it be possible to also allow `my_happy_event_name` as an alias (only to be used by user-facing apis), assuming no conflicts?

Oops, I somehow overlooked this question. Doing it like it's implemented now keeps it consistent with other plugin extended things, e.g. templates, `div`-ids, permissions in 1.4.0.dev and so on. Allowing customized event names without prefix will only cause issues and frustration down the road if I need a conflicting event name.

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel