OctoPrint/OctoPrint #2966: [Request] More distinction between PrintCancelled and PrintFailed

tjjfvi, 2018-12-22:
**Is your feature request related to a problem? Please describe.**
Right now, when one cancels a print, it fires both a PrintCancelled event and a PrintFailed event. Sometimes, this is what you want (e.g. you saw that the print headcrabbed via the webcam), and sometimes you don't it to be a "failed" print (e.g. you simply selected the wrong file).

**Describe the solution you'd like**
When you cancel a print, give an option to not also send a PrintFailed event. Also, when a PrintFailed event was triggered by a cancel, add a payload prop (like `cancel=true`), so plugins can ignore that if desired and not have to track if there was a PrintCancelled event immediately before.

**Describe alternatives you've considered**
I haven't thought of any alternatives.

**Additional context**
Add any other context or screenshots about the feature request here.

tjjfvi, 2018-12-27:
I just saw that the PrintFailed payload has a `reason` prop that is either `cancelled` or `failed`. _\*facepalm\*_
But my checkbox on cancelling a print still applies.

foosel, 2019-01-10:
Hm... I'm not sure about that. By definition, a print that didn't complete failed. It didn't complete after all. And the rare cases where it would make sense to diverge from that general rule of thumb I fear don't justify adding an additional popup (or try to figure out how to cram in a checkbox next to the cancel button) to allow the user override this.

Wouldn't it be better to just allow marking a print as failed/non failed, at any time?

tjjfvi, 2019-01-10:
In my mind, the checkbox would be in the confirm dialog, but thinking about that now I realize that may not be feasible / may not be a great solution.

I like the idea of checking a print as succeeded or failed (and maybe not printed as well) at any time, though.

foosel, 2019-01-11:
The confirmation dialog can be disabled, so that would be tricky UX wise (what if I want the control over fail marking but don't want the confirmation?).

Maybe the best solution here is really just manually marking prints as failed/successful. Will have to find a solution though what to do about the events that get triggered in such cases - so far they are inherently bound to a job that just was done. And I think I'd like to keep it that way honestly.

tjjfvi, 2019-01-11:
Would the marking print as failed / succeeded be a thing for a plugin?

foosel, 2019-01-18:
I just realized that it's not as easy after all since failed/successful is based on the [print log](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/filemanager/__init__.py#L604), see [here](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/server/api/files.py#L215-L231).

So a solution would have to either

  * remove the last history entry, effectively resetting back to the earlier one or - if there is none - a black status
  * introduce some form of "manual toggle" history entry that gets ignored by the other statistics running on this data but is used for the last state

The first option could be done by a plugin, but maybe a clean option defined on the filemanager would be preferable to have here first. The second option feels like it adds way too much overhead just for a simple color mark up in the frontend.