OctoPrint/OctoPrint #2971: Remember Me not working in 1.3.10

melgish, 2018-12-24:
#### What were you doing?
After upgrading one of two instances to 1.3.10 discovered that upgraded instance requires login despite repeatedly checking remember-me button. 

Tracked problem to line 20 of src/octoprint/plugins/forcelogin/static/js/main.js.   Line should read:
`var remember = rememberElement.prop("checked");`  because rememberElement is a jQuery instance.

1. Browse to Octoprint
2. If Logged in, Logout, Clear history, or delete cookies
3. Close and Re-open Browser
4. Login to Octoprint. Be sure "Rememeber Me" is checked.
5. Close and Re-open Browser

#### What did you expect to happen?
User auto logged in using cookie credentials.

#### What happened instead?
User forced to reenter name and password.

#### Did the same happen when running OctoPrint in safe mode?
Not tested. Problem found while troubleshooting.

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
Not Applicable

#### Browser and version of browser, operating system running browser
Chrome and Firefox. Both latest at time of submit.

#### Link to octoprint.log
Not Applicable.  Problem is in browser code.

#### Link to contents of terminal tab or serial.log
Not Applicable.  Problem is in browser code.

#### Link to contents of Javascript console in the browser
Not applicable. Nothing relevant to issue logged.

#### Screenshot(s)/video(s) showing the problem:
Not applicable.

I have read the FAQ.

tedder, 2018-12-25:
Ahhh. So it isn't just me.

foosel, 2019-01-18:
Should be fixed by #2973 and will be part of 1.3.11

Scorexy, 2019-05-14:
1.3.11 is released and remember me still does not work :(

foosel, 2019-05-15:
It works here 🤔

Can you please provide *exact reproduction steps*?

Scorexy, 2019-05-15:
> It works here 
> 
> Can you please provide _exact reproduction steps_?

**to access directly to TouchUI as octopi boots up, I either disable access control:**

accessControl:
  enabled: false

**or add localNetworks in the accessControl:**

accessControl:
  localNetworks:
  - 127.0.0.0/8
  - 192.168.56.0/24
  - ::1/128

**now I have updated to 1.3.11 and removed this line to check if remember me butten is now active:**
  - ::1/128

**I put my login data and ticked the remember me box and rebooted, but all the fields are still empty as they were with 1.3.10.**

foosel, 2019-05-16:
"Remember me" isn't supposed to fill in the username and password for you, it's supposed to not even make you see the login screen and automatically log you in. It sets a cookie which will serve as a auth token. Of course, your browser needs to persist that cookie. The question is: does it?

Can you reproduce this only on TouchUI or also on a regular browser accessing your instance? I'm not sure if TouchUI uses it's own authentication mechanism in its local install or if it falls back to OctoPrint's stuff - @BillyBlaze can you clarify?

Also, just to check - you said you disable ACL *or* enable local logins. I don't see how you'd ever even see a login screen from OctoPrint if ACL is disabled, so I'm taking it we are talking about a local autologin setup here that's intentionally crippled to remove ::1/128?

BillyBlaze, 2019-05-16:
Running TouchUI directly on a Pi with the bootloader will disable the cache of Chromium, so every time you restart the browser means you have to re-login, this is why we now suggest to setup `autoLoginAs`.

However, if dont want this, simple do the following
1. Edit TouchUI default:
    `sudo nano /etc/default/touchui`

2. Add at the end of the file the following variable:
    `CONFIG_DIR=/home/pi/.config/chromium/`

3. Save (ctr-o, enter, ctr-x) and restart TouchUI (you do need to login once more)

Scorexy, 2019-05-16:
> "Remember me" isn't supposed to fill in the username and password for you, it's supposed to not even make you see the login screen and automatically log you in. It sets a cookie which will serve as [...]

I don't see the login screen when ACL is disabled or using local login.

However I was wondering why my login data kept disappearing even when 'remember me' box was checked in previous login.  I thought this was a bug, but now BillyBlaze cleared it out.

Thank you all

foosel, 2019-05-17:
Closing this then, thanks everyone 😊

closed 2019-05-17 by foosel