OctoPrint/OctoPrint #2974: [Request] Enable/Disable Webcam/Timelapse functionality instead of clearing a string.

thantik, 2018-12-26:
I have read the FAQ.

Enabling and disabling the webcam and timelapse features in OctoPrint should be simplified from: Make these two text boxes empty.  If I want to enable them again, I've gotta go somewhere external and look up what strings need to be placed back in those boxes to make the webcam work again.

Please either break these out into a plugin I can enable/disable, so that I can disable this functionality or simply add an enabled/disabled flag so I can disable the functionality without losing the strings needed for the webcam software to pick up the image.

This is just one of those 'death by a thousand paper cuts' things.  You've actually run into a bug regarding this functionality because of the odd way it's handled: https://github.com/foosel/OctoPrint/issues/2206

foosel, 2019-01-18:
Implemented by the above commit, pushed to `maintenance`, soon `devel`, to be released with 1.3.11.

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel