OctoPrint/OctoPrint #2975: Error updating to 1.3.10

OutsourcedGuru, 2018-12-27:
## (Just documenting a failed 1.3.10 upgrade attempt)

#### What were you doing?

* Began with new installation of 2018-04-18-octopi-stretch-lite-0.15.1.img.
* Connected to wi-fi as expected
* Went through Setup Wizard from web interface: disable access control & connectivity check, enable blacklisting, skip, basic printer profile
* See errors in the pasted popup dialog seen below

#### What did you expect to happen?
As usual (and this worked earlier today on another install), I expected the 1.3.10 upgrade to be happy and restart.

#### What happened instead?

```
Update Failed!
The update did not finish successfully. Please consult octoprint.log and plugin_softwareupdate_console.log for details.
```

#### Did the same happen when running OctoPrint in safe mode?
No plugins installed yet, vanilla install.

#### Work-around noted
* Settings -> Software Update -> Check for Update -> Update

"Update Successful, Restarting!"

#### Version of OctoPrint
1.3.8 from 0.15.1 OctoPi image dated 2018-04-18

#### Operating System running OctoPrint
Raspbian Stretch Lite, see above

#### Printer model & used firmware incl. version
None yet

#### Browser and version of browser, operating system running browser
macOS Mojave 10.14.2, Safari 12.0.2

#### Contents of update attempt

```
Updating, please wait.
/home/pi/oprint/bin/python2 -m pip install https://github.com/foosel/OctoPrint/archive/1.3.10.zip
Collecting https://github.com/foosel/OctoPrint/archive/1.3.10.zip
Downloading https://github.com/foosel/OctoPrint/archive/1.3.10.zip
Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Collecting tornado==4.5.3 (from OctoPrint==1.3.10)
Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(104, 'Connection reset by peer'))': /packages/e3/7b/e29ab3d51c8df66922fea216e2bddfcb6430fb29620e5165b16a216e0d3c/tornado-4.5.3.tar.gz
Downloading https://files.pythonhosted.org/packages/e3/7b/e29ab3d51c8df66922fea216e2bddfcb6430fb29620e5165b16a216e0d3c/tornado-4.5.3.tar.gz (484kB)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: markdown<2.7,>=2.6.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Collecting sarge==0.1.5post0 (from OctoPrint==1.3.10)
Downloading https://files.pythonhosted.org/packages/c4/2b/deaaacf4af3f9c45c48be04a6a48fec60515fb34dafda9fe61ecd2c5e4cc/sarge-0.1.5.post0.tar.gz
Requirement already satisfied: netifaces<0.11,>=0.10.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: rsa<3.3,>=3.2.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Collecting pkginfo<1.5,>=1.4.2 (from OctoPrint==1.3.10)
Downloading https://files.pythonhosted.org/packages/a3/fe/f32a48d48f40a7209be9825fba2566cab92364787cf37de2e08300dd6ce7/pkginfo-1.4.2-py2.py3-none-any.whl
Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.3.10)
Collecting psutil<5.5,>=5.4.6 (from OctoPrint==1.3.10)
Downloading https://files.pythonhosted.org/packages/e3/58/0eae6e4466e5abf779d7e2b71fac7fba5f59e00ea36ddb3ed690419ccb0f/psutil-5.4.8.tar.gz (422kB)
Exception:
Traceback (most recent call last):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
status = self.run(options, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 335, in run
wb.build(autobuilding=True)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/wheel.py", line 749, in build
self.requirement_set.prepare_files(self.finder)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
ignore_dependencies=self.ignore_dependencies))
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
session=self.session, hashes=hashes)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
hashes=hashes
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
hashes)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 882, in _download_http_url
_download_url(resp, link, content_file, hashes)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 603, in _download_url
hashes.check_against_chunks(downloaded_chunks)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/hashes.py", line 46, in check_against_chunks
for chunk in chunks:
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 571, in written_chunks
for chunk in chunks:
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/ui.py", line 139, in iter
for x in it:
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 560, in resp_read
decode_content=False):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/urllib3/response.py", line 436, in stream
data = self.read(amt=amt, decode_content=decode_content)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/urllib3/response.py", line 401, in read
raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/urllib3/response.py", line 320, in _error_catcher
raise ProtocolError('Connection broken: %r' % e, e)
ProtocolError: ("Connection broken: error(104, 'Connection reset by peer')", error(104, 'Connection reset by peer'))
You are using pip version 9.0.3, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
```

#### Link to octoprint.log
[octoprint-8.log](https://github.com/foosel/OctoPrint/files/2713334/octoprint-8.log)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

FanDjango, 2018-12-28:
I can see in you log:
```
2018-12-27 12:19:31,225 - octoprint.plugins.softwareupdate - ERROR - Update of octoprint can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 909, in _perform_update
    update_result = updater.perform_update(target, populated_check, target_version, log_cb=self._log, online=online)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 110, in perform_update
    raise exceptions.UpdateError("Error while executing pip install", (stdout, stderr))
UpdateError
```

I am not sure but if I remember correctly, I experienced this once too a while ago. 
Try the following:
Logon on to that system and as root, do a
```pip install --upgrade pip```
and try the OctoPrint upgrade again

OutsourcedGuru, 2018-12-28:
I note that from the `octoprint.log` throughout the 1.3.10 upgrade process we see the following for `pip` itself. Not knowing any better, I would think that it wanted an update since `/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py` threw an exception. Back in 1.3.8, line 110 of this file was the generic handler for "any of the following egg files had a problem". Maybe this then wasn't a problem with pip trying to update itself but just some unknown egg...?

## Before:
```
|  python:
|    pip: 9.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
```

## After first failed attempt:
```
|  python:
|    pip: 9.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
```

## After second (seemingly-successful) attempt:
```
|  python:
|    pip: 9.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
```

foosel, 2019-01-10:
```
raise ProtocolError('Connection broken: %r' % e, e)
ProtocolError: ("Connection broken: error(104, 'Connection reset by peer')", error(104, 'Connection reset by peer'))
```

This is the reason for the temporary upgrade failure. `pip` ran into a connection issue with the Python package index while trying to fetch dependencies. This is nothing that OctoPrint can do anything about, if there are network hiccups there are network hiccups. As you noticed, a second attempt to install it worked fine (because there were no hiccups anymore).

closed 2019-01-10 by foosel

OutsourcedGuru, 2019-01-10:
Like I said, the upgrade usually works great. I've gone through the same routine numerous times both in testing and in various activities.

I guessed that you'd like to see the "smoking gun" behind one of these upgrade failures, that's all.