OctoPrint/OctoPrint #2976: Autologin in fresh install not working in 1.3.10 with OctoPi 0.15.1

han9988, 2018-12-29:
#### What were you doing?
Configured Autologin on my two Raspberry Pis: 
Pi1: OctoPi 0.14.0 and OctoPrint 1.3.10 - autologin works as expected
Pi2: OctoPi 0.15.1 and OctoPrint 1.3.10 - autologin NOT working

nano ~/.octoprint/config.yaml
accessControl:
  salt: ++++++private++++++++++++++++++
  autologinLocal: true
  autologinAs: dimitri
  localNetworks:
  - 127.0.0.0/8
  - 192.168.0.0/24

#### What did you expect to happen?
Autologin sets configured user and logs into web interface. 

#### What happened instead?
On Pi1 everything works as expected
On Pi2 sometimes just the login is presented, sometimes the layout template not filled but never autologin

#### Did the same happen when running OctoPrint in safe mode?
Did not try since it is working in Pi1

#### Version of OctoPrint
Pi1: OctoPi 0.14.0 and OctoPrint 1.3.10 - autologin works as expected
Pi2: OctoPi 0.15.1 and OctoPrint 1.3.10 - autologin NOT working

#### Operating System running OctoPrint
Pi1: Raspberry Pi b+Version 8
Pi2: Raspberry Pi b+ Version 9.6

#### Printer model & used firmware incl. version
Not needed

#### Browser and version of browser, operating system running browser
Not Browser related

#### Link to octoprint.log
https://gist.github.com/han9988/33f24aedf82205e0e2ef16fcdf781a5b

#### Link to contents of terminal tab or serial.log
Not needed

#### Link to contents of Javascript console in the browser
This page includes a password or credit card input in a non-secure context. A warning has been added to the URL bar. For more information, see https://goo.gl/zmWq3m.

#### Screenshot(s)/video(s) showing the problem:
Not needed

I have read the FAQ.
Shure - Why would I take the time to describe this in detail and not investigate first???

foosel, 2019-01-10:
Could you try [this](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/266#issuecomment-447810686)?

> Shure - Why would I take the time to describe this in detail and not investigate first???

You would be surprised...

han9988, 2019-01-10:
Hi foosel,
I already tried adding IPv6 temporarily before issuing my problem here and it didn't help.
Since I do have a working instance on another Pi as mentioned, I did the following now:
Reconfigured localNetworks and added ::1/128 to the not working raspberry pi.
Restarted the raspberry pi and went straight to the octoprint.log to compare the two raspberry pi‘s.

Both log files look identical to the last line:

Raspberry Pi-1 (Autologin working)
2019-01-10 20:43:31,185 - octoprint.server.preemptive_cache - INFO - ... done in 29.23s

Raspberry Pi-2 (Autologin not working)
2019-01-10 20:46:41,150 - octoprint.server.preemptive_cache - INFO - ... done in 29.32s

Now open Browser and navigate to Octoprint:

Raspberry Pi-1 (Autologin working)
2019-01-10 20:44:41,828 - octoprint.server.util.flask - INFO - Passively logging in user dimitri from 192.168.0.26 via autologin
2019-01-10 20:44:44,470 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.0.26
2019-01-10 20:44:44,600 - octoprint.server.util.flask - INFO - Passively logging in user dimitri from 192.168.0.26
2019-01-10 20:44:47,130 - octoprint.server.util.sockjs - INFO - User dimitri logged in on the socket from client 192.168.0.26
-End
Result in Browser: Autologin

Raspberry Pi-2 (Autologin not working)
2019-01-10 20:49:50,668 - tornado.access - WARNING - 404 GET /favicon.ico (::ffff:192.168.0.26) 18.25ms
2019-01-10 20:49:50,708 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.26
-End
Result in Browser: Login-Dialog presented

Raspberry Pi-2 (Autologin not working) without the ::1/128 entry in config.yaml:
2019-01-10 21:12:58,643 - tornado.access - WARNING - 404 GET /favicon.ico (::ffff:192.168.0.26) 16.08ms
2019-01-10 21:12:58,686 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.26
2019-01-10 21:13:00,364 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.26

Additional info:
Raspberry Pi-1 is connected via Ethernet cable
Raspberry Pi-2 is connected via WLAN

Why the hybrid IP notation? I didn't configured this.
Also odd: The Client connection closed message without the ::1/128 entry in config.yaml and no further entry when with the ::1/128
Hope this infos help.

han9988, 2019-01-12:
status:awaiting information

foosel, 2019-01-14:
> Why the hybrid IP notation? I didn't configured this.

As far as I understand that's because the newer OctoPi and hence Raspbian supports V6 whereas your old one didn't. In both cases OctoPrint itself only listens on V4 (OctoPi sets `--host 127.0.0.1` on the startup parameters, unless you've changed that), but on one the underlying OS supports V6 and on the other it doesn't. Have to look further into why it's reported that way though.

> Also odd: The Client connection closed message without the ::1/128 entry in config.yaml and no further entry when with the ::1/128

That is indeed odd.

Something to try: What happens if you add `::ffff:192.168.0.0/120` to `localNetworks`?

> Howto enable detailed logging at this point?

[Logging can be configured as explained here](http://docs.octoprint.org/en/master/configuration/logging_yaml.html) but there's not more detailed logging in the autologin component I think that would help use here, so if the above doesn't work I'll have to prep a branch for you to try.

> Possible file right problem, maybe?

I honestly don't think so.

han9988, 2019-01-14:
I added ::ffff:192.168.0.0/120 to localNetworks and this is the solution.
Autologin works now as expected.

My working config.yaml:

accessControl:
  autologinAs: username
  autologinLocal: true
  localNetworks:
  - 127.0.0.0/8
  - 192.168.0.0/24
  - ::1/128
  - ::ffff:192.168.0.0/120
  salt: xxx

Based on the previous experience I kept the ::1/128 entry. Although it is not neccessary for the solution.
Maybe the solution to the Touch UI problem is similar, just to add ::ffff:127.0.0.1/128 ?
Anyway, glad we found this without too much of a fuzz. ;-)

closed 2019-01-14 by han9988

foosel, 2019-01-14:
Perfect! I also just pushed a commit (link see above) that will automatically add the v6 version of configured v4 networks to the list during processing.

> Maybe the solution to the Touch UI problem is similar, just to add ::ffff:127.0.0.1/128 ?

Rather `::ffff:127.0.0.1/104`, but as far as I can see the problems in that ticket are related to requests originating from `::1`, not `::ffff:127.0.0.1`. Still, I'll mention it.