OctoPrint/OctoPrint #2979: OctoPrint disconnects in the middle of the print (Not using a Pi)

jsbursik, 2018-12-30:
#### What were you doing?

Printing a 10mm cube, uploaded the file to octoprint-venv, clicked print.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

Print a 10mm cube

#### What happened instead?

Print failed approx. halfway through (6 min job, fails ~3 minutes in)

#### Did the same happen when running OctoPrint in safe mode?

Can't start in safemod, using Arch Linux with octoprint-venv from the AUR

#### Version of OctoPrint

1.3.10

#### Operating System running OctoPrint

Arch Linux 4.19.12-arch1-1-ARCH

#### Printer model & used firmware incl. version

Printer is a converted Da vinci Jr. 1.0, Marlin 1.1.9

#### Browser and version of browser, operating system running browser

Chrome Version 71.0.3578.98 (Official Build) (64-bit), Windows 10 Pro (64-bit) 

#### Link to octoprint.log

https://gist.github.com/jsbursik/fd87e813e27f48711b625231669b6440

#### Link to contents of terminal tab or serial.log

Just turned serial log, and refreshed the page, going to have to fail a print again to grab this.

#### Link to contents of Javascript console in the browser

Refreshed page, need to fail again.

#### Screenshot(s)/video(s) showing the problem:

Unnecessary.

I have read the FAQ.

I would like to add some extra information: octoprint is being run in a systemd service provided by the AUR package, it has permission to the group "uucp" as well as "tty" (even though $ stat /dev/ttyUSB0 returns only the group uucp).  I tried tripping the safe mode flag by doing "$ octoprint safemode" then restarting the service, no dice. This issue has been reported before, I am very much aware, but in all of those, it's always been "Swap USB cables, check power supply" The MKS GEN that is running marlin has it's own power supply, and the server running octoprint has quite enough power. Also, the cable being used is the ONLY cable that works with the board, as I've tried others. It is shielded and I've had no random disconnects except for while printing. I've tried this print 3 times, it fails at what I assume is the same point, every time. I have noticed that if I watch the webcam for less than a minute it blacks out until I reload the page.

jsbursik, 2018-12-30:
Here is what I have done since:
- I disconnected the webcams 
- Disconnected an LED strip in the case (was running off of the MKS GEN board, but I noticed flickering from power draw)
- Set the timeouts to be longer, like, A LOT longer (30 seconds in most cases)

Ran another print of the cube, and it successfully printed, plugged the LED strip back in, print failed at the very top layer with the same error (the one in the gists in my post that I just added). Haven't tested with the webcam being added back in yet. I also want to test printing from SD card (have to install one first) to see if OctoPrint loses connection at any point regardless of where the gcode is coming from.

tedder, 2019-01-01:
What type of device/server is this running on?

jsbursik, 2019-01-01:
It's a computer running Arch Linux (Latest) on an AMD FX-6300 CPU, Don't remember the motherboard specifics, but it's the RS780L Chipset.

foosel, 2019-01-10:
```
2018-12-30 04:43:19,346 - Send: N2124 G0 F7200 X76.827 Y71.17*66
2018-12-30 04:43:19,983 - Recv: ok
2018-12-30 04:43:19,985 - Send: N2125 G1 F900 X78.829 Y73.172 E212.87237*13
2018-12-30 04:43:20,175 - Recv:  T:205.04 /205.00 @:79
2018-12-30 04:43:20,236 - Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2605
2018-12-30 04:43:20,237 - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
2018-12-30 04:43:20,243 - Changing monitoring state from "Printing" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2605)"
2018-12-30 04:43:20,250 - Connection closed, closing down monitor
```

I know you don't want to hear this but - there's nothing else that I can recommend here *but* looking into cabling and/or power issues here. The problem is that the serial connection breaks away under OctoPrint at a layer below where it can do anything. It just vanishes. Like *poof*. There's nothing I can do to prevent this (that I'm so far aware of, and I'd be very surprised if there was anything that could be done in general).

The reason I recommend to check cabling and power is because in the past, these kind of situations highly correlated with bad cabling or bad power and fixing both made sudden disconnects go away.

If disconnecting the LED stips makes it harder to reproduce or even unreproducibly, it could be the power draw of the LED strip interfering with things. But it's impossible to say from afar. This is really more of an environmental issue, possibly hardware, possibly firmware, possibly OS, than a bug in OctoPrint. And I can't do anything about it, which is why I'll close this ticket. Comments are still possible, but please understand that I really can't do anything about this in OctoPrint and so it doesn't make sense to keep this open.

closed 2019-01-10 by foosel