OctoPrint/OctoPrint #2981: Request: Connect after 'EEPROM datasize error'

marchyman, 2019-01-01:
I updated firmware on my printer (a modified CR-10S if anyone cares).   I changed some configuration values (number of ABL probes and ABL probe locations) which are stored in EEPROM.  As a result the EEPROM did not validation.  That would not normally be an issue as my first steps after the printer restarted would have been something like `M502; M206 Y-4.00; M851 Z-1.80; M500; M117 Firmware Updated` which is what my printer likes.

However, because of the error OctoPrint won't connect.

```
2019-01-01 00:14:46,960 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: EEPROM datasize error.
| Last lines in terminal:
| Connecting to: /dev/ttyUSB0
| Changing monitoring state from "Offline" to "Opening serial port"
| Connected to: Serial<id=0x6f6428d0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
| Changing monitoring state from "Opening serial port" to "Connecting"
| Send: N0 M110 N0*125
| Recv: start
| Send: N0 M110 N0*125
| Recv: echo: External Reset
| Recv: Marlin bugfix-1.1.x
| Recv: 
| Recv: echo: Last Updated: 2018-07-31 | Author: (SNAFU CR-10S)
| Recv: echo:Compiled: Dec 31 2018
| Recv: echo: Free Memory: 2844  PlannerBufferBytes: 1408
| Recv: Error:EEPROM datasize error.

```

That meant that instead of jumping into the terminal tab and fixing the issue I had to string a USB cable from a computer that was barely close enough to the printer and run Pronterface to connect and issue the commands needed to rebuild the EEPROM.

**Describe the solution you'd like**
Do not disconnect after EEPROM errors.  Perhaps disable printing, but please allow terminal access.

FanDjango, 2019-01-01:
I experienced that as well so then I changed the error behaviour from "stop printing and disconnect" to just "stop printing" (i.e. no disconnect). 
Then I can still rectify the situation in the octopring terminal window.
You can do that in the settings for the serial connection.

marchyman, 2019-01-05:
Thank you FanDjango.  Seems I never noticed the Behavior tab.   I'll close this non-issue.

closed 2019-01-05 by marchyman