OctoPrint/OctoPrint #2982: [Request] User activity logging

kerog777, 2019-01-01:
In an environment with multiple users, there's no way to tell who started an active print.  So if something goes wrong and you need to cancel the print, there's no way to tell who should be informed.

I'd like the infrastructure to be put in place so that people could write plugins around this sort of thing.  It was suggested in the discussion forum (Topic: WhoStartedIt Plugin) that this could be done by adding login and print start/stop activity to the octoprint log.  Some plugins I could see:

* Who started the current print
* Display detailed activity log (file printed, person starting, person stopping, time started, time stopped, etc)
* Log a message when canceling a print to explain why.

Useful associated features would be
* Add contact phone number and email address to account info
* Add generic tagged logging capability so that, for instance, when changing filament one could add a log note with the brand, color, and amount remaining on the spool that could then be displayed for other users to see

tedder, 2019-01-01:
When Gina gets back from vacation hopefully she has some insight on the user part, which seems perfect to add to the [PrintStarted event](http://docs.octoprint.org/en/master/events/#printing), making it easy for a plugin to show the detailed pieces.

The filament parts are probably best managed with the FilamentManager plugin.

As far as adding the contact info, that feels like it's headed down the path of [Zawinski's Law](http://www.catb.org/jargon/html/Z/Zawinskis-Law.html); if you're running that intense of a printing lab you should be able to hire a software geek or subsidize some Octoprint development.

foosel, 2019-01-10:
> which seems perfect to add to the [PrintStarted event](http://docs.octoprint.org/en/master/events/#printing), making it easy for a plugin to show the detailed pieces.

I agree. And I'm in fact slightly confused right now that this isn't already coming with the username attached, since that was attached that to print jobs a couple versions ago and also displays in the UI. Oversight I guess :)

> As far as adding the contact info, that feels like it's headed down the path of [Zawinski's Law](http://www.catb.org/jargon/html/Z/Zawinskis-Law.html); if you're running that intense of a printing lab you should be able to hire a software geek or subsidize some Octoprint development.

I agree.

foosel, 2019-01-16:
Print job events now contain the `owner` of a job (= the user who selected the file or ultimately started it) and the `user` who triggered the action (start, pause, resume, cancel). Commit is pushed to `maintenance`, will be merged to `devel` ASAP and be released as part of 1.3.11.

Documentation is currently being built and should hopefully soon be up [here](http://docs.octoprint.org/en/maintenance/events/index.html#printing).

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel