OctoPrint/OctoPrint #2983: [Request] find file string input box automaticly clearing after click on the result.

njozsef, 2019-01-02:
Hi Developers!!!

My request little complicated. Sorry for worst English.
Find files (folder) on octoprint,
after selected (clicked) folder, not cleared string the input box automatically.
if files on targeted (clicked) folder not contain pattern, not listed. For example: search string: tensioner, contain file on tensioner foler is "nut.stl": 
Only visible all file on folder, when deleted all letters on search box lot of pressing the backspace.
Just a clear button.
This is little 'comfort' problem, not major.
short example screenshot video(25 sec): https://www.youtube.com/watch?v=sRLBt0MJtKw
Thx

foosel, 2019-01-10:
I understood you want to have a clear button in the search box, is that correct?

Automatic clearing after you have found what you are looking for won't be possible because it isn't possible for OctoPrint to determine that.

foosel, 2019-01-16:
Implemented by the above commit, available on `maintenance`, soon `devel`, to be released as part of 1.3.11.

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel