OctoPrint/OctoPrint #2985: When print sent and heated. It homes then does nothing

rotaryfurball, 2019-01-02:
#### What were you doing?

1. Press print. Wait for bed and nozzle to heat up
2. ...
3. ...

#### What did you expect to happen?
For the print to start printing
#### What happened instead?
It homes then does nothing
#### Did the same happen when running OctoPrint in safe mode?
Yep

#### Version of OctoPrint
OctoPrint 1.3.10 running on OctoPi 0.14.0

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

tedder, 2019-01-02:
Please upload your octoprint log and serial log or *we can't help you*.

FanDjango, 2019-01-20:
Oh, I get this kind of problem often. As a matter of fact I will open an issue soon on this, but I want to collect more info and also check what is really happening - pyserial is flaky, in my opinion.
Look at the screenshots, solely after a fresh connect to the printer, it is a matter of luck for octoprint to actually see a vaild "ok" to the "M113 S29" at the end of its specific initialisation sequence.
[image]

[image]

[image]
This also happens after a "Firmware Update", of course, = a reconnect, and takes some measures to fix before you can start a print.

Like I said, I have read ALL the other threads about serial communications problems and am following this topic. There is definitively a problem here and connection via minicom do NOT reflect these symptoms.

foosel, 2019-03-08:
Closing due to lack of response & logs.

closed 2019-03-08 by foosel