OctoPrint/OctoPrint #2987: ABL Expert Cooldown "Feature" - Read this if ur Printer stops heating after BL G29

reloxx13, 2019-01-04:
This ABL Expert Plugin in i cuzing alot of troubles for ppls out there.
If users have a g29 in ther gcode start code, by default this plugin will stop the heating and the print wont start.

**@guys who got the problem:** u dont need to add G29 in ur startcode. use octoprint/terminal/lcd to start the probe, save config (M500) and load config (M501) after the probe to load the new mesh data.
to enable the mesh, put M420 S1 in ur startcode after G28 home.
alternative enable RESTORE_LEVELING_AFTER_G28 in ur marlin configuration

@devs
if possible only enable this if the probe is started due octoprint.
if its in the gcode dont stop the heating.

alternative: atleast write in terminal why the heat is shutting down "ABL Expert cooldown nozzle/bed after probing now"

there are help post cuz of this everyday on the fb octoprint help group.
this is creating more troubles than it helps ppls.

(default is checked)
[image]

enable this plugin as default was a bad idea.
track with ur statistics how many users got a bltouch and would even use this.

 this orange warning is also bugged, my printer is connected and currently printing.

PS2: and yes, its not neccessary to G29 before every print, but still alot ppls do it and i think almost every utube guide tells that.

#2862

tedder, 2019-01-04:
This should be posted in the ABL plugin repo, not the main Octoprint one, right?

reloxx13, 2019-01-04:
i dunno since this is a default enabled plugin now

tedder, 2019-01-04:
ABL is not in the default distribution.

reloxx13, 2019-01-04:
r u sure? some fb guys told me after the latest update this plugin was installed and the heating stopped.

jubaleth, 2019-01-04:
Hi @reloxx13, @tedder is right, this isn't a bundled plugin. This issue should be brought up on the plugin's repo (https://framagit.org/razer/Octoprint_ABL_Expert).

closed 2019-01-04 by jubaleth

razerraz, 2019-01-20:
I'm the maintainer of this plugin, so you can blame me.
Please first read my post on the issue in framagit : 
https://framagit.org/razer/Octoprint_ABL_Expert/issues/3

I'm sorry for this late response, I've issue with email notifications with framagit and was not aware of that.

And please : **don't use this plugin if you don't need mesh correction**, since it's the main purpose of it. Seems now clear this thing is somewhere buggy, and if you don't have any reason to install it and get troubles it generate, just uninstall it.

I will fix the issues mentioned here soon

razerraz, 2019-01-20:
Just release v0.2 that should solve this issue along others.
Another bug affect software update and force to uninstall previous version and reinstall it. This issue should be fixed too.