OctoPrint/OctoPrint #2988: [Request] Tell pip to stop confusing the noobs with upgrade suggestions

OutsourcedGuru, 2019-01-04:
**Is your feature request related to a problem? Please describe.**
Each and every `pip install` activity is displayed for users both in the Settings-related popup and in the logs/command line.

A lot of the users seem to think that any warning is a bad thing. The warning/message in question is related to the pip suggestion that pip itself ought to be upgraded. Some of the time this results in users asking on the forum what they should do. Some of the time users then take it upon themself to attempt to "fix" their pip version, getting into trouble.

**Describe the solution you'd like**
`pip install --disable-pip-version-check _whatever_`

...or edit the `$VIRTUAL_ENV/pip.conf` with this information so that manual commands don't throw the "error".

**Describe alternatives you've considered**
We could attempt to distribute the information in the documentation and forum but it's not usually read first, to be honest.

**Additional context**
They see something similar to this:
https://i.stack.imgur.com/Sedia.jpg

foosel, 2019-01-10:
Yeah, I fear it really has come to this. Been pondering it for a while, but even though there's an FAQ entry about it, people still get too confused by it, so that needs to be done I guess.

tedder, 2019-01-10:
There were what, 2-3 separate threads on it over the break? I thought it was silly at first to do this, but obv I'm wrong.

OutsourcedGuru, 2019-01-10:
I know, they're just wearing us out with things-they-shouldn't-be-doing, to be honest. Granted, Christmas is a big time for buying new 3D printers and toys.

foosel, 2019-01-11:
I actually setup an FAQ entry for that a couple months back, but I guess it's not news that people don't read.

foosel, 2019-01-11:
There you go. Done through the above commit, to be part of 1.3.11.

OutsourcedGuru, 2019-01-11:
Thanks.

People are so event-driven these days. It's based upon the [constant upgrade](https://outsourcedguru.wordpress.com/2017/09/14/experience-is-simply-the-name-we-give-our-mistakes/) mentality out there.

I have several Macs, iPads, iPhones and they all nag me to upgrade this, that and the other thing. We're being programmed that we MUST take all updates or face some unknown consequences.

OutsourcedGuru, 2019-02-04:
[Updating OctoPi / not a directory](https://discourse.octoprint.org/t/updating-octopi-not-a-directory/7288?u=outsourcedguru)

> I saw the blog entry about OctoPi 0.15.0 and 0.16.0, so I want to update my OctoPi.
When using the command: pip install --upgrade pip
I got error messages. So I used following commands to update pip (I thought so)...

🤕

OutsourcedGuru, 2019-03-02:
For what it's worth, I note that [this page](https://docs.octoprint.org/en/master/development/environment.html) suggests that the developer upgrade their pip. I'm not sure whether or not that this can get someone into trouble.

[image]

qqgg231, 2019-03-30:
I think I have related issue. I use PrusaPrint image. It is my first day, so I can start with new image if this can't be solved easily

I did not remove any lines from below, so you can see when things stopped working

1. Manual extension installation was working (OctoPrint plugin manager coudn't access internet, but bash doesn't mind )
```
pi@octopi:~ $ pip install "https://github.com/OctoPrint/OctoPrint-FirmwareUpdate                                                                             r/archive/master.zip"
-bash: pip: command not found
pi@octopi:~ $ ~/oprint/bin/pip install "https://github.com/OctoPrint/OctoPrint-F                                                                             irmwareUpdater/archive/master.zip"
Collecting https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/master                                                                             .zip
  Downloading https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/mas                                                                             ter.zip
     \ 471kB 3.3MB/s
Requirement already satisfied: OctoPrint in ./oprint/lib/python2.7/site-packages                                                                              (from OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: flask<0.11,>=0.10.1 in ./oprint/lib/python2.7/sit                                                                             e-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in ./oprint/lib/python2.7/site                                                                             -packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: regex!=2018.11.6 in ./oprint/lib/python2.7/site-p                                                                             ackages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in ./oprint/lib/python2.7/si                                                                             te-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: tornado==4.5.3 in ./oprint/lib/python2.7/site-pac                                                                             kages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: PyYAML<3.11,>=3.10 in ./oprint/lib/python2.7/site                                                                             -packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in ./oprint/lib/python2.                                                                             7/site-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in ./oprint/lib/pytho                                                                             n2.7/site-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in ./oprint/lib/python2.7/                                                                             site-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in ./oprint/lib/python2.                                                                             7/site-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: markdown<2.7,>=2.6.9 in ./oprint/lib/python2.7/si                                                                             te-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: pyserial<3.5,>=3.4 in ./oprint/lib/python2.7/site                                                                             -packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in ./oprint/lib/python2.7/si                                                                             te-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in ./oprint/lib/python2.7/si                                                                             te-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: sarge==0.1.5post0 in ./oprint/lib/python2.7/site-                                                                             packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: netifaces<0.11,>=0.10.6 in ./oprint/lib/python2.7                                                                             /site-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: pylru<1.1,>=1.0.9 in ./oprint/lib/python2.7/site-                                                                             packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: rsa<3.3,>=3.2.3 in ./oprint/lib/python2.7/site-pa                                                                             ckages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
Requirement already satisfied: pkginfo<1.5,>=1.4.2 in ./oprint/lib/python2.7/sit                                                                             e-packages (from OctoPrint->OctoPrint-FirmwareUpdater==1.1.0)
[... cut ...]
pi@octopi:~ $ ~/oprint/bin/pip install "https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/archive/master.zip"
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/archive/master.zip
Exception:
...
...
...
```

foosel, 2019-04-01:
@qqgg231 that doesn't look like a related issue but rather an install that after several partial installs of something got broken. Might be fixable, but the easiest is probably to reflash since I honestly don't know *what* broke.

foosel, 2019-04-04:
And now I've also pushed a change which will suppress pip 19.0+'s newest message of "OMG PY2.7 IS EOL!!1!" which has also already caused confusion among users. I get why they are doing this, but I'd only prefer if they were a bit less annoying about it.

Anyhow... above commit is ready for 1.3.11rc2

foosel, 2019-04-04:
Correction... 1.3.11rc3 because I fscked up a merge. Sigh.

b-morgan, 2019-04-08:
So what version of pip should we be using? 

If the pip we have is not the pip we need, can you automagically fix it a) if it is older? b) if it is newer?

foosel, 2019-04-09:
If it really needs updating, yes, I'll find a way. But rather than forcing updates of pip, for the most part OctoPrint just is built to work with pretty much every version. The only cases where that doesn't apply are broken versions (which OctoPrint will detect) and version which are too old to interact with the current iteration of the official python package index (which OctoPrint will also tell you).

Long story short: in most cases you don't have to use any other version that what came with your python install and/or image, and if that changes I'll find a way to solve it ideally automatically.

The problem with updating pip - which is why I don't have it set to automatically update by default - is that newer versions sometimes introduce bad breaking changes in how it's used, and that can really break things in the field at unforeseen times in a way that needs fiddling on the command line to get back into working order.

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel