OctoPrint/OctoPrint #2992: [Bug] Restore of backup on docker-mounted --basedir volume due to rename logic

hgross, 2019-01-07:
#### What were you doing?
Trying to restore a backup.

1. Create a backup using the backup mechanism and download it.
2. Start a second octoprint instance with the exact same versions/codebase of plugins etc.
3. Upload the backup zip file there
4. See error

```

Uploading backup, this can take a while. Please wait...
 
Stelle Sicherung wieder her...
 
Unpacking backup to /tmp/tmpYhY_SP...
Unpacked
Renaming /data/octorpint-data to /data/octorpint-data.bck...
Moving /tmp/tmpYhY_SP/basedir to /data/octorpint-data...
Error while restoring config data
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.10-py2.7.egg/octoprint/plugins/backup/__init__.py", line 828, in _restore_backup
    os.rename(basedir_extracted, basedir)
OSError: [Errno 18] Invalid cross-device link
Rolling back old config data
 
Wiederherstellung fehlgeschlagen! Prüfe die obige Ausgabe sowie octoprint.log für mögliche Gründe dafür.
Removing temporary unpacked folder
Removing temporary zip

```
I started octoprint using this script inside a docker container on a raspberry pi 3b+:
```
#!/bin/bash

# ensure that octoprint data will be on 2nd+ level folder to not make restore of backups fail trying to rename the docker mapped volume /data
mkdir -p /data/octorpint-data

# start Octoprint
octoprint --iknowwhatimdoing --port=8080 --basedir /data/octorpint-data
```
As you can see in the script, I circumvented a related issue - the restore mechanism calls `os.rename` on a mounted volume that cannot be renamed resulting in `OSError: [Errno 18] Invalid cross-device link`. This is most probably related to docker, because I mount the /data directory from the host machine, as it is a common way to handle state with docker containers. See https://stackoverflow.com/questions/42392600/oserror-errno-18-invalid-cross-device-link for workaround suggestions. For me it was fine to move the data dir to `/data/octorpint-data` instead of `/data` to work around that issue. Sadly the `os.rename(a,b)` call failed again. Maybe you can just print out a warning in this case (Errno 18) or switch to a copy, then delete behaviour (here https://github.com/foosel/OctoPrint/blob/112a8b9c8bf7f5895f6a29dbe3d98ea4d9e0f400/src/octoprint/plugins/backup/__init__.py#L828)? Or maybe shutil will help working around that (as mentioned in the linked SO-link). 

#### What did you expect to happen?
Restore of the state.

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
1.3.10 (both instances)

#### Operating System running OctoPrint
raspian (not OctoPi)

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log
https://gist.github.com/hgross/41fa4272b9549260c5a29a87b8a5ac03

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

hgross, 2019-01-07:
The `shutil.move` alternative works well with the docker mounts and even falls back to the currently implemented `os.rename` implementation if appropriate (see https://docs.python.org/2/library/shutil.html#shutil.move). 

```
Uploading backup, this can take a while. Please wait...
 
Stelle Sicherung wieder her...
 
Unpacking backup to /tmp/tmpI5tPoN...
Unpacked
Renaming /data/octorpint-data to /data/octorpint-data.bck...
Moving /tmp/tmpI5tPoN/basedir to /data/octorpint-data...
Removing temporary unpacked folder
Removing temporary zip
 
Wiederherstellung erfolgreich! Der Server wird jetzt neu gestartet!
```

The server did not restart on its own and I had some problems with not existing caches though (did not save the logs unfortunately).
After a manual restart the preemptive cache generation on octoprint startup fixed that problem automatically. 

I created a PR.

foosel, 2019-01-11:
PR has been merged to `maintenance` and will be part of 1.3.11 (and also be merged to `devel` ASAP). Thanks!

PR #2993 (2019-01-11): Fix #2992 (restore mechanism on docker volumes)

  * [x] Your changes are not possible to do through a plugin and relevant
    to a large audience (ideally all users of OctoPrint)
  * [x] If your changes are large or otherwise disruptive: You have
    made sure your changes don't interfere with current development by
    talking it through with the maintainers, e.g. through a
    Brainstorming ticket
  * [x] Your PR targets OctoPrint's devel branch, or maintenance if it's
    a bug fix for an issue present in the current stable version (no PRs
    against master or anything else please)
  * [x [... cut]
commit 4728a1a5 (2019-01-11): Merge pull request #2993 from hgross/fix/restore-on-docker-volumes

Fix #2992 (restore mechanism on docker volumes)
closed 2019-05-14 by foosel