OctoPrint/OctoPrint #2994: Ocotoprint doesn not add M530 S1 command at print start

ZhenyaKa, 2019-01-09:
For example Repetier-server add this command internally.
M530 S1 at begin of print
and 
M530 S0 at end of print

This need for thermal protection. If not add printer shut heaters off after 30 mins of printing.
Of cource you can said that need to add it manually in script fields, but other USB hosts I no need to do that.

ZhenyaKa, 2019-01-09:
https://github.com/MKFirmware/MarlinKimbra/issues/190

Here the problem of stop heating

foosel, 2019-01-10:
You are free to modify your [GCODE scripts](http://docs.octoprint.org/en/master/features/gcode_scripts.html) however it best suits your setup. I can't just blindly add commands that are not available on all firmware variants out there and might trigger god knows what by default though.

closed 2019-01-10 by foosel

ZhenyaKa, 2019-01-10:
Okay, I added it already, but a lot of people will print bad 30 mins thing. 

Repetier also have this command
https://github.com/repetier/Repetier-Firmware/blob/master/doc/capabilities.txt