OctoPrint/OctoPrint #2995: [Request][Metaticket] Better managing of disk storage and gracious handling of running out of space

foosel, 2019-01-10:
This is a meta ticket to track the various requests regarding a better handling of low disk space scenarios in various situations:

  * https://github.com/foosel/OctoPrint/issues/2312 (run out during update)
  * https://github.com/foosel/OctoPrint/issues/655 (run out during upload)
  * https://github.com/foosel/OctoPrint/issues/2964 (run out during timelapse recording)
  * https://github.com/foosel/OctoPrint/issues/1030 (warning about running out soon)

This ticket is supposed to collect all these for easier reference.

dwery, 2021-03-26:
That would be nice to have, stray logs in my pi filled up the root partition and I would have noticed it sooner if I had a warning in the UI

LazeMSS, 2021-03-31:
This is also on the list:
[Request] Better error handling when disk is full during file uploads #655