OctoPrint/OctoPrint #3000: Timelaspe disabled itself

iandouglas, 2019-01-13:
#### What were you doing?

I had enabled the timelapse feature on OctoPrint and got some great still-capture videos of my prints. Last night while doing a print I checked the settings panel and it was disabled ("timelapse mode" was "off"). I'm the only person with access to OctoPrint on my home network. This isn't the first time I've seen this happen.

I don't recall rebooting my Pi 3 at all, but wouldn't expect a setting like that to toggle because of a reboot anyway.

#### What did you expect to happen?

I expected that once I enabled Timelapse mode that I'd always get generated movies unless I manually shut it back off.

#### What happened instead?

Uncontrolled by me, Timelapse Mode shut itself off.

#### Did the same happen when running OctoPrint in safe mode?

Haven't tried. I print a LOT on my printer and rely on OctoPrint so I've never run it in Safe Mode to see if my printing still works.

#### Version of OctoPrint

1.3.10

#### Operating System running OctoPrint

OctoPi 0.15.1

#### Printer model & used firmware incl. version

Prusa MK 3, firmware 3.5.0

#### Browser and version of browser, operating system running browser

Chrome 71, Windows 10

#### Link to octoprint.log
#### Link to contents of terminal tab or serial.log

there's nothing in my logs about "timelapse" and I don't know when it toggled off (some time after January 3rd, my logs only go back to the 6th it seems)

#### FAQ

Yes, I have read the FAQ.

tedder, 2019-01-13:
This sounds entirely unreproducable.

iandouglas, 2019-01-14:
Maybe, but it's incredibly annoying when I touch no settings whatsoever and
yet timelapse is randomly shut off.

foosel, 2019-01-14:
You say this isn't the first time this has happened. Have other settings toggled or is it just the timelapse one?

Also please provide an `octoprint.log`. It's not only interesting due to entries relating to the specific issue but also gives required insight into your setup that could lead to ideas on what's happening here.

foosel, 2019-03-08:
Closing due to lack of response & logs.

closed 2019-03-08 by foosel