OctoPrint/OctoPrint #3001: sd printing => auto cancel after start (randomly)

devildant, 2019-01-14:
Hello,
I encounter a random problem with printing via sd, sometimes when I run a print from the sd card (since octoprint), after heating and home, the temperature drops to 0 on the printer, however the print continues. octoprint side that shows me cancelling and operational.

i try change SD, flash... same. 

the problem occurs randomly :(

i see that inside serial log : 
2019-01-14 17:41:32,403 - Send: M27
2019-01-14 17:41:32,430 - Recv: Not SD printing

#### Version of OctoPrint

OctoPrint 1.3.10

#### Operating System running OctoPrint

octopi

#### Printer model & used firmware incl. version

cr-10s, marlin 1.1.9 th3d

#### Browser and version of browser, operating system running browser

chrome

#### Link to octoprint.log

2019-01-14 17:39:16,330 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-01-14 17:39:16,491 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2019-01-14 17:39:16,495 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2019-01-14 17:39:21,035 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-01-14 17:39:21,137 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin TH3D U1.R2.7"
2019-01-14 17:39:21,149 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2019-01-14 17:39:21,277 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2019-01-14 17:39:21,379 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2019-01-14 17:39:46,022 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-01-14 17:40:16,076 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-01-14 17:40:46,131 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-01-14 17:41:16,185 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-01-14 17:41:19,667 - octoprint.filemanager.analysis - INFO - Starting analysis of local:caissson_tirroir.gcode
2019-01-14 17:41:19,672 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/caissson_tirroir.gcode
2019-01-14 17:41:29,506 - octoprint.filemanager.analysis - INFO - Analysis of entry local:caissson_tirroir.gcode finished, needed 9.84s
2019-01-14 17:41:32,367 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Printing from SD"
2019-01-14 17:41:32,438 - octoprint.util.comm - INFO - Changing monitoring state from "Printing from SD" to "Cancelling"
2019-01-14 17:41:32,628 - octoprint.plugins.shutdownprinter - INFO - Print started
2019-01-14 17:41:32,649 - octoprint.plugins.shutdownprinter - INFO - Print cancelled
2019-01-14 17:41:46,240 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-01-14 17:42:16,294 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-01-14 17:42:46,347 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-01-14 17:43:16,399 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-01-14 17:43:35,184 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-01-14 17:43:46,454 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-01-14 17:44:07,689 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"

#### Link to contents of terminal tab or serial.log
serial.log : 
2019-01-14 17:41:30,276 - Send: M21
2019-01-14 17:41:30,827 - Recv: echo:SD card ok
2019-01-14 17:41:30,853 - Recv: ok
2019-01-14 17:41:30,857 - Send: M20
2019-01-14 17:41:30,875 - Recv: Begin file list
2019-01-14 17:41:30,877 - Recv: CAISSS~1.GCO 1438703
2019-01-14 17:41:30,890 - Recv: End file list
2019-01-14 17:41:30,895 - Recv: ok
2019-01-14 17:41:31,497 - Recv:  T:26.14 /0.00 B:28.91 /0.00 @:0 B@:0
2019-01-14 17:41:32,294 - Send: M23 /caisss~1.gco
2019-01-14 17:41:32,327 - Recv: echo:Now fresh file: /caisss~1.gco
2019-01-14 17:41:32,328 - Recv: File opened: caisss~1.gco Size: 1438703
2019-01-14 17:41:32,331 - Recv: File selected
2019-01-14 17:41:32,363 - Changing monitoring state from "Operational" to "Printing from SD"
2019-01-14 17:41:32,396 - Recv: ok
2019-01-14 17:41:32,403 - Send: M27
2019-01-14 17:41:32,430 - Recv: Not SD printing
2019-01-14 17:41:32,437 - Changing monitoring state from "Printing from SD" to "Cancelling"
2019-01-14 17:41:32,615 - Recv: ok
2019-01-14 17:41:32,620 - Send: M23 /caisss~1.gco
2019-01-14 17:41:32,635 - Recv: echo:Now doing file: /caisss~1.gco
2019-01-14 17:41:32,647 - Recv: File opened: caisss~1.gco Size: 1438703
2019-01-14 17:41:32,653 - Recv: File selected
2019-01-14 17:41:32,657 - Recv: ok
2019-01-14 17:41:32,664 - Send: M24
2019-01-14 17:41:32,679 - Recv: ok
2019-01-14 17:41:32,683 - Send: M108
2019-01-14 17:41:32,706 - Recv: ok T:26.14 /0.00 B:28.97 /75.00 @:0 B@:0
2019-01-14 17:41:32,711 - Send: M84
2019-01-14 17:41:32,715 - Recv:  T:26.14 /0.00 B:28.97 /75.00 @:0 B@:0 W:?
2019-01-14 17:41:33,492 - Recv:  T:25.91 /0.00 B:28.97 /75.00 @:0 B@:127
2019-01-14 17:41:33,808 - Recv:  T:26.14 /0.00 B:28.91 /75.00 @:0 B@:127 W:?

devildant, 2019-01-14:
from my point of view it is as if sending the command M27 was sometimes too fast, suddenly the printer responds that it does not print via SD

devildant, 2019-01-14:
I found this issue that seems to be a similar problem (https://github.com/foosel/OctoPrint/issues/2910), but I do not know what it is that "Redeem" 
ps I inserted logs in the description

foosel, 2019-01-15:
Looks like a race condition in the SD start procedure, I'll take a look.

devildant, 2019-01-15:
I did not give the complete logs for you to avoid searching.
but if necessary I kept them.
if i can help do not hesitate

tedder, 2019-01-15:
Is your Pi overclocked, @devildant?

devildant, 2019-01-15:
Hello @tedder , 
no, I tried with two different pi (3b and 3b +), it's the same (random bug)

devildant, 2019-01-15:
what is complicated is that it can take 5 min or more days to reproduce the bug

foosel, 2019-01-17:
So this is definitely a race condition, and I also know where it comes from, why it happens and how to fix it. Currently testing the fix.

devildant, 2019-01-17:
as I have to re-test the  the hook "SD card upload" , I will make a stone two shot, and also tested this fix, for against I would make a return to the minimum in a week (unless I notice the bug immediately: p), as it is random I prefer to wait a little :)
thanks for the fix.
I will make a return as soon as possible

> Funnily enough I also observed a case where the SD print auto cancelled, then an M27 was sent that returned a progress status and OctoPrint switched back to printing.

for me, the M27 command was not returned and, as option M108 was sent, the temperature was set to 0 for the bed and the hotend

devildant, 2019-01-17:
I installed the maintenance branch, I will use the printer to see if I have the problem again.
I will come back to this question as soon as possible

foosel, 2019-01-25:
The trick is that the `M27` can still happen before the `M24` but a report of `Not SD printing` will no longer cause an issue while the print is still in Starting state. So that looks good. The `M108` option shouldn't really affect the outcome here regardless of its setting.

devildant, 2019-01-25:
yes probably, as long as the heating has not started, no problem for the M108 command.
it looks good, I tested all week and I have not encountered any problem.
I looked at the logs yesterday to make my return.

foosel, 2019-01-25:
I guess we can really call it solved then :) Thanks for the rigorous testing! 👍

devildant, 2019-01-25:
I always prefer to test a maximum to make sure that it works well, to avoid coming back again and again on the same problem.
even though I think it would have taken several weeks to be 100% sure.
but a week without problems I think it's good.
thanks for the work :)

commit a48e069a (2019-01-17): Fix race condition on start of SD print

If an M27 to query SD status came after the change of state to printing
but before the M24 to actually start the print on the printer's side the
firmware would report "Not SD printing" and OctoPrint would consequently
cancel the just start job right away.

This patch introduces a new "Starting" state that will be active while
preparing a print job (in case of SD: selecting the file, setting file
position & finally starting the print). The comm layer treats as
nearly identically to the Printing state, the only difference is in the
treatment of received " [... cut]
closed 2019-01-25 by devildant