OctoPrint/OctoPrint #3002: Octoprint disconnects on error even if set to stay connected in case of error

MasterPIC, 2019-01-15:
Hi,
by simulating an heating failure on my printer I found that Octoprint disconnects.
It reports:

> Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Heating failed, system stopped! Heater_ID: bed - Printer halted. kill() called!

I set Octoprint to stay connected to the printer in case of error (Error handling panel).
Is it a bug?
Should I simply install OctoPrint-FirmwareErrorHandler plugin?

In the meantime I changed the firmware to make the printer triggering a specific action before going to report the error...

Thanks
Armando

foosel, 2019-01-15:
> Is it a bug?

No. The error your printer reported was an error where it's not possible to stay connected as the firmware has entered a full blown error state that makes it unresponsive ("kill() called"). The only thing OctoPrint could do here is keep the serial open, but it would not be able to do anything with the printer anymore until you manually reset it.

Turning this into a request to clarify this in the settings dialog.

MasterPIC, 2019-01-15:
I edited my printer firmware by adding

`SERIAL_ECHOLNPGM("//action:" ACTION_ON_KILL);`

after the lines used by the printer to report the error but **still before the kill call**.

```
void Temperature::_temp_error(const int8_t heater, PGM_P const serial_msg, PGM_P const lcd_msg) {
  static bool killed = false;
  if (IsRunning()) {
    SERIAL_ERROR_START();
    serialprintPGM(serial_msg);
    SERIAL_ECHOPGM(MSG_STOPPED_HEATER);
    if (heater >= 0) SERIAL_ECHOLN((int)heater); else SERIAL_ECHOLNPGM(MSG_HEATER_BED);
    SERIAL_ECHOLNPGM("//action:" ACTION_ON_KILL);
  }
  #if DISABLED(BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE)
    if (!killed) {
      Running = false;
      killed = true;
      kill(lcd_msg);
    }
    else
      disable_all_heaters(); // paranoia
  #endif
}
```
By Octoprint terminal, with Octoprint in safe mode, I see that **Octoprint detects the action request**...

Maybe it's an issue related only to Action commands plugin?
If I move SERIAL_ECHOLNPGM("//action:" ACTION_ON_KILL); before the code used to report the error, the action is executed.

Thanks again
Armando

foosel, 2019-01-15:
> By Octoprint terminal, with Octoprint in safe mode, I see that **Octoprint detects the action request**.
> But the action is not executed, with Octoprint not in safe mode...

If something works in safe mode and doesn't work in normal mode, a third party plugin that doesn't come bundled with OctoPrint is to blame.

MasterPIC, 2019-01-15:
I totally agree with you.
I assumed the action request was not detected... Then I made other tests and it seems the action is not executed if the request occurs after the error has been reported...

closed 2019-01-15 by MasterPIC