OctoPrint/OctoPrint #3006: GCode Visualizer still running even if disabled

nforceroh, 2019-01-18:
#### What were you doing?
gcode visualizer runs even if disabled in settings which causes Raspberry PI Zero to come to a crawl after octoprint is restarted.
```
pi@delta01:~ $ ps -ef|grep octoprint
pi         342     1 57 12:23 ?        00:05:11 /home/pi/oprint/bin/python2 /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000
pi         862   342 33 12:24 ?        00:02:28 /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 /home/pi/.octoprint/uploads/Body_Cover.gcode
 a few minutes later
pi        3400   342 42 12:38 ?        00:01:18 /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 /home/pi/.octoprint/uploads/Wheel_extruder.gcode
````
#### What did you expect to happen?
gcode visualizer should not run if disabled

#### What happened instead?
gcode visualizer  is running

#### Version of OctoPrint
OctoPrint 1.3.10 running on OctoPi 0.15.1

#### Printer model & used firmware incl. version
Marlin 2.x

#### Link to octoprint.log
```2019-01-18 12:23:57,672 - octoprint.startup - INFO - ******************************************************************************
2019-01-18 12:23:57,682 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2019-01-18 12:23:57,685 - octoprint.startup - INFO - ******************************************************************************
2019-01-18 12:23:57,957 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2019-01-18 12:24:01,501 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2019-01-18 12:24:01,753 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-01-18 12:24:17,538 - octoprint.plugin.core - INFO - Found 30 plugin(s) providing 30 mixin implementations, 44 hook handlers
2019-01-18 12:24:18,098 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-01-18 12:24:18,407 - octoprint.server - INFO - Intermediary server started
2019-01-18 12:24:18,416 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-01-18 12:24:23,070 - octoprint.plugin.core - INFO - Found 30 plugin(s) providing 30 mixin implementations, 44 hook handlers
2019-01-18 12:24:23,399 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2019-01-18 12:24:27,082 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2019-01-18 12:24:27,261 - octoprint.plugins.cancelobject - INFO - No beforegcode defined
2019-01-18 12:24:27,279 - octoprint.plugins.cancelobject - INFO - No aftergcode defined
2019-01-18 12:24:29,390 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2019-01-18 12:24:41,935 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2019-01-18 12:24:54,565 - octoprint.util.pip - INFO - Version of pip is 9.0.3
2019-01-18 12:24:54,575 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-01-18 12:24:54,583 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-01-18 12:24:54,602 - octoprint.plugin.core - INFO - Initialized 30 plugin implementation(s)
2019-01-18 12:24:54,782 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2019-01-18 12:24:54,789 - octoprint.plugins.psucontrol - INFO - Using internal tracking for PSU on/off state.
2019-01-18 12:24:54,796 - octoprint.plugins.psucontrol - INFO - Running RPi.GPIO version 0.6.5
2019-01-18 12:24:54,804 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2019-01-18 12:24:54,811 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 19
2019-01-18 12:24:54,894 - octoprint.plugin.core - INFO - 30 plugin(s) registered with the system:
2019-01-18 12:24:55,026 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
2019-01-18 12:24:55,132 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2019-01-18 12:24:55,255 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2019-01-18 12:24:57,943 - octoprint.server - INFO - Shutting down intermediary server...
2019-01-18 12:24:58,151 - octoprint.server - INFO - Intermediary server shut down
2019-01-18 12:24:58,168 - octoprint.events - INFO - Processing startup event, this is our first event
2019-01-18 12:24:58,179 - octoprint.events - INFO - Adding 1 events to queue that were held back before startup event
2019-01-18 12:24:58,203 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2019-01-18 12:24:58,313 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2019-01-18 12:24:58,847 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance "Delta01"' for _http._tcp
2019-01-18 12:24:59,093 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance "Delta01"' for _octoprint._tcp
2019-01-18 12:24:59,226 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "Delta01" for SSDP
2019-01-18 12:24:59,701 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2019-01-18 12:24:59,770 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Body_Cover.gcode
2019-01-18 12:24:59,863 - octoprint.plugins.bedlevelvisualizer - INFO - OctoPrint-BedLevelVisualizer loaded!
2019-01-18 12:24:59,930 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 /home/pi/.octoprint/uploads/Body_Cover.gcode
2019-01-18 12:25:00,651 - octoprint.plugins.navbartemp - INFO - Broadcom detected
2019-01-18 12:25:01,220 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2019-01-18 12:25:03,996 - octoprint.filemanager - INFO - Added 2 items from storage type "local" to analysis queue
2019-01-18 12:25:23,276 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2019-01-18 12:25:25,340 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
[... 6 lines cut ...]
2019-01-18 12:30:07,825 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2019-01-18 12:30:12,456 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial port"
2019-01-18 12:30:12,792 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2019-01-18 12:30:12,930 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-01-18 12:30:13,082 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2019-01-18 12:30:13,210 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-01-18 12:30:13,397 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin bugfix-2.0.x (Github)"
2019-01-18 12:30:13,479 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2019-01-18 12:30:13,530 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2019-01-18 12:31:36,828 - octoprint.server.preemptive_cache - INFO - ... done in 348.14s
2019-01-18 12:31:36,865 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.253.171/'}
2019-01-18 12:34:21,927 - octoprint.server.preemptive_cache - INFO - ... done in 165.06s
2019-01-18 12:38:16,962 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Body_Cover.gcode finished, needed 797.19s
2019-01-18 12:38:18,103 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Wheel_extruder.gcode
2019-01-18 12:38:18,138 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 /home/pi/.octoprint/uploads/Wheel_extruder.gcode
2019-01-18 12:39:18,145 - octoprint.server.heartbeat - INFO - Server heartbeat 
```

config.yaml settings
````pi@delta01:~/.octoprint $ more config.yaml 
feature: {}
gcodeViewer:
  enabled: false
plugins:
```

nforceroh, 2019-01-18:
updated template

foosel, 2019-01-18:
What you disabled isn't the GCODE visualiser or viewer in the front-end but rather the GCODE analysis that's needed for estimating filament usage and print time that runs in the backend. Not a bug. Possibly a feature request to be able to disable analysis? If so please explain the motivation behind it.

foosel, 2019-01-21:
> that analysis seems to run for every gcode file in the upload directory if you reboot, restart octoprint or when you upload a new file

The analysis only runs for files for which it hasn't run yet and hence no analysis results are available.

> As a gcode streamer, the RPi Zero is very capable but I usually will kill that process or wait a few minute before I start printing because it does affect GCODE streaming to the printer.

This process being quite CPU intense is why OctoPrint aborts the analysis process prior to starting a print. It will never run parallel to streaming a job on an unmodified OctoPrint for this exact reason.

> Maybe a simple regex to capture and present that data as the filament length in the UI, or have the option to use slicer info, ignore or force Octoprint's analysis

As @jneilliii pointed out, there are plugins out there that utilize that information.

zman97211, 2019-03-22:
Perhaps an alternative would be to specify the process priority of the analysis or limit the CPU usage. The OctoPrint web page fails to server properly sometimes due to this, which means I can't even get a print to start (which is what pauses the analysis).

ForsakenNGS, 2019-03-29:
> The analysis only runs for files for which it hasn't run yet and hence no analysis results are available.
> This process being quite CPU intense is why OctoPrint aborts the analysis process prior to starting a print. It will never run parallel to streaming a job on an unmodified OctoPrint for this exact reason.

I've run into this issue slowing down my print extremely. Admitted my setup is a little bit of an edge-case.
I'm running a beagleboard with two(!) octoprint instances simultaneous for my two 3d printers attached to it. Usually it works pretty nice, but upon reboot or when uploading big gcode files the octoprint analysis pretty much completely blocks my cpu and will NOT stop when starting a print (because of the two seperate octoprint instances)

My only way to get around that right now is a) Cleaning up my uploads before doing a restart and b) Checking (h)top after uploading a big gcode file.

Honestly I've never used the information gathered from the analysis, so I would just disable it when given the option. Alternatively both suggestions by nforceroh and zman97211 seem reasonable to me.

Alissonverd, 2019-04-11:
[ just an idea, do not crucify me ]

We could take the initiative to standardize the information provided slicers?
Today they provide the same data, but with different names, units and in different places.
We can request these changes for the most commonly used slicers and thus begin a standardization of the supplied header data.
If the analysis also addresses checking for a corrupted file, a checksum may be one of the data provided on header data.

foosel, 2019-04-24:
@zman97211

> Perhaps an alternative would be to specify the process priority of the analysis or limit the CPU usage.

Good idea. Would need to be done in a platform agnostic way though (OctoPrint is *not* Linux-only). [This looks like a potential approach](https://stackoverflow.com/questions/1023038/change-process-priority-in-python-cross-platform).

@ForsakenNGS 

> Honestly I've never used the information gathered from the analysis

You might not have used it personally, but OctoPrint uses it internally, e.g. for print time estimation. If you think it's inaccurate with the metadata analysis results available, wait till you see how it performs if that data is *not* available.

However, I'm open to adding a switch for that somewhere in the advanced options or maybe just in the config. The last thing I want to do though is make disabling that an readily available and visible option, because I can already smell the tickets pouring in from people disabling it and then wondering why they no longer get filament usage and print time estimates out of their instances...

@Alissonverd

> [ just an idea, do not crucify me ]

Not going to crucify anyone, regardless of the quality of the idea, and this one is even a very good one. The problem though is push back from the community. In my experience, trying to standardize even something as crucial for operation as the communication protocol to the printer only ends in enormous push back and outright hostility. If someone gets some people around a table, makes a suggestion for a centralized format with a proper spec, I'd be happy to implement that and then see if slicers do as well. But I won't do that coordination or specification part myself, I've already been burnt a couple of times in efforts like this and frankly have enough grey hair already.

sergio-daniels, 2019-09-09:
I never had this problem since I set up octopi 0.16 with 1.3.11, I get this uge cpu load for the analysis of the gcode. My pi model B+ V1.2 is unresponsive through the web interface. Once I started the print through telegram bot the cpu dropt for the python2 anaysis task

kakou-fr, 2019-10-16:
same problem, and during a print, i uploaded some gcode, the cpu overheat, octopi decreased the CPU freq, and my print was destroyed.

buddydvd, 2019-11-02:
I'd like to disable it for a different reason.

My concern is related to #1488.

I help manage multiple printers and I'd like to have all associated Octoprint instances access GCODE files from one shared folder on the network. The fact that Octoprint recursively analyzes each GCODE file in the `uploads` folder makes this workflow problematic.

I understand the problems associated with different instances writing to the same `.metadata.json` file. That problem can potentially be avoided with the use of OverlayFS (i.e., `lowerdir=/mnt/shared/`). However, copying new files into the shared folder will cause OctoPrint to analyze each unseen files. When the shared folder introduces many files, each OctoPrint instance will load each unseen file and analyze it. It would be great if OctoPrint offer the ability to not do that.

Can OctoPrint's plugin system support the creation a plugin that offers a storage implementation that omits metadata creation / GCODE analysis?

hashashin, 2019-11-02:
To avoid the gcode analysis in the octoprint machine I use a combination of https://github.com/malnvenshorn/OctoPrint-MetadataPreprocessor and https://github.com/hashashin/OctoWatcher but the second can be replaced with a post-slice script on many slicer programs.

Salud.

buddydvd, 2019-11-09:
I created a plugin to disable GCODE analysis.

You can find it here: https://github.com/buddydvd/octoprint-disable-gcode-analysis

Please check it out and let me know if you see anything wrong.

foosel, 2019-11-13:
There are now a bunch of related options added by #3337:

[image]

Merged and to pushed out with the next release.

foosel, 2020-03-04:
1.4.0 has been released.

closed 2020-03-04 by foosel

notlebowski, 2020-12-01:
This "only when idle" doesn't work from a practical point of view.
Yes, it will only start when it's not printing, but this also means it will slow down the boot process.
In my case it takes hours to start the Web Interface

I now set it to "never", but I would prefer to set it back to "when idling" if it would also include a half hour sleep before it starts rendering those files. 

[image]

cp2004, 2020-12-01:
>In my case it takes hours to start the Web Interface

Must be adding a lot of files to the folders while the server is not running, that it needs to analyse then. It only analyses files once, not every boot. More details are needed if you think this is a problem, such as logs...

notlebowski, 2020-12-01:
It's still busy rendering those 123 gcode-files after 2 hours.
I still think the "only when idle" should be more "intelligent", but if it indeed does it only once I can live with it.

If it has finished them all I will issue a service restart and see what happens....

```
2020-12-01 16:05:31,440 - octoprint.filemanager.analysis - INFO - Analysis of entry local:plate_11jpr-255.gcode finished, needed 20.49s
2020-12-01 16:05:32,446 - octoprint.filemanager.analysis - INFO - Starting analysis of local:plate_11jpr.gcode
2020-12-01 16:05:32,449 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=18000 --speed-y=18000 --max-t=10 --throttle=0.01 --throttle-lines=100 /home/pi/.octoprint/uploads/plate_11jpr.gcode
2020-12-01 16:05:53,449 - octoprint.filemanager.analysis - INFO - Analysis of entry local:plate_11jpr.gcode finished, needed 21.00s
2020-12-01 16:05:54,453 - octoprint.filemanager.analysis - INFO - Starting analysis of local:plate_2.gcode
2020-12-01 16:05:54,455 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=18000 --speed-y=18000 --max-t=10 --throttle=0.01 --throttle-lines=100 /home/pi/.octoprint/uploads/plate_2.gcode
2020-12-01 16:06:26,250 - octoprint.filemanager.analysis - INFO - Analysis of entry local:plate_2.gcode finished, needed 31.80s
2020-12-01 16:06:27,261 - octoprint.filemanager.analysis - INFO - Starting analysis of local:raft-RadioLab_Taung_Child_Skull_via_shootdigital_2014.05.15.gcode

```

notlebowski, 2020-12-01:
Yep it starts doing them all again.
Here is part of the log where it seems to "initialize" the metadata that was gathered.

The name of the file is not mentioned nor why it should initialize again.

```
2020-12-01 17:03:12,992 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-12-01 17:03:13,018 - octoprint.server - INFO - Intermediary server started
2020-12-01 17:03:13,046 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/                                    uploads...
2020-12-01 17:03:37,848 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.

```

foosel, 2020-12-01:
`2020-12-01 17:03:13,046 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/                                    uploads...`

This just means it's scanning if there's anything to analyze. If you don't see individual files, nothing gets processed.

notlebowski, 2020-12-01:
> This just means it's scanning if there's anything to analyze. If you don't see individual files, nothing gets processed.

In my first sentence I stated that it's doing them all again.

Another 3 hours of scanning:

```
2020-12-01 17:54:35,794 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-12-01 17:55:27,946 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 2752}
2020-12-01 17:55:57,825 - octoprint.filemanager.analysis - INFO - Analysis of entry local:M_RadioLab_Taung_Child_Skull_via_shootdigital_2014.05.gcode finished, needed 926.26s
2020-12-01 17:55:58,833 - octoprint.filemanager.analysis - INFO - Starting analysis of local:M_Tika_Alpaca_2_Dec.gcode
2020-12-01 17:55:58,835 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=18000 --speed-y=18000 --max-t=10 --throttle=0.01 --throttle-lines=100 /home/pi/.octoprint/uploads/M_Tika_Alpaca_2_Dec.gcode

```

notlebowski, 2020-12-01:
@foosel 
I'm running Octoprint 1.5 on a Rpi 3 which was an image install.
Please share your thoughts about the reasons it does these renderings each time Octoprint starts.

foosel, 2020-12-01:
The only reason I can see this happening is if it cannot write the analysis results back to the metadata files, or the file changes don't persist across reboots. Both indicate a dead or dying file system and/or SD card.

notlebowski, 2020-12-01:
Where are those metadata files so I can check?
Why isn't it logging that it can't write to those files? That's what logging is all about.

A faulty SD-Card is not likely as it would have shown in the various updates it went through as this is a fresh install on a new Pi with a new SD-Card. I'm not saying that a new SD-Card can't be faulty, but it is impossible that it would only do this for these files.

If I know where that metadata is supposed to go, I can check that.

foosel, 2020-12-01:
It can only log that it can't write if it knows it can't write. If the file system claims the write succeeded, it can't detect this. Even fresh SD cards can be faulty, we have seen this before in the community. I'm not saying this is necessarily the issue here, but it would explain the behaviour you are seeing that goes against what should actually be happening according to the implementation.

Metadata is stored in the uploads folder and all its subfolders in a file called `.metadata.json`.

notlebowski, 2020-12-01:
I had a .metadata.json file in there which I deleted yesterday. It was of my old setup and it still had the "2017" filestamp.

A new ,metadata.json is created, but it contains only metadata of 1 file. and an empty json of a 2nd one

```
cat uploads/.metadata.json
{
    "Base_plate.gco": {
        "analysis": {
            "printingArea": {
                "maxX": 119.021,
                "maxY": 94.403,
                "maxZ": 27.44,
                "minX": 32.975,
                "minY": 43.299,
                "minZ": 0.24
            },
            "dimensions": {
                "depth": 51.104000000000006,
                "height": 27.200000000000003,
                "width": 86.04599999999999
            },
            "estimatedPrintTime": 1369.4446376032774,
            "filament": {
                "tool0": {
                    "length": 826.05995,
                    "volume": 0.0
                }
            }
        }
    },
    "Battery_Holder_for_65MM_quads.gco": {}

Note the opening and immediately closing parenthesis.
That's not a filesystem error when writing, but another kind of logical error.

notlebowski, 2020-12-02:
I mounted my old SD-Card again with a USB-stick and copied the .metadata.json to the uploads folder.

```
sudo mount /dev/sda2 /mnt/usb/
cp -p  /mnt/usb/home/pi/.octoprint/uploads/.metadata.json ./uploads/

```
I moved Base_plate.gco to a folder I named   ,octoprint/uploads.2018
I counted the "gco" entries

```
pi@octopi:~/.octoprint $ grep -c gco ./uploads/.metadata.json
151
pi@octopi:~/.octoprint $ sudo service octoprint start
```
I waited a moment...
It turns out it removed that entry
```
pi@octopi:~/.octoprint $ grep -c gco ./uploads/.metadata.json
150

```
That's looking good.

Moved the file back restarted the service and now it's 151 again.
It seems it has difficulties to do them all from scratch and I can't put my finger on it.

abc2006, 2021-04-17:
Same Problem for me. Octoprint was running fine (and idle) for days. Now i had to reboot and it analyses all files at uploads. (again?). 
I will wait until its finished and then reboot again.

abc2006, 2021-04-17:
rebooted and octoprint starts analyzing again

cp2004, 2021-04-17:
Please open a bug report if you think you have found a bug.