OctoPrint/OctoPrint #3007: Duet WiFi support

Patrik-svobodik, 2019-01-19:
Hi, I tried to use octoprint with duet wifi, but I cannot load files or upload them to the SD card, octoprint does not detect when I start a print from DWC or PanelDue, it looks like it does not support RRF right, could you look into it? I love OctoPrint, but to get the best performance of the duet you have to print from the SD directly. There are a lot of clients for OctoPrint, so that is why I want to use it

foosel, 2019-01-21:
> it looks like it does not support RRF right, could you look into it?

I really need a lot more information in order to even answer that question. Most importantly a `serial.log` from when you try to do that stuff that doesn't work.

Patrik-svobodik, 2019-01-22:
> I'd be happy to look into it, but as I said, I need more info for that to
> even start, first of all a serial.log to see what the printer is even
> returning.

Ok, will send when I will get home

foosel, 2019-03-08:
Closing due to lack of response & logs.

closed 2019-03-08 by foosel

imrj, 2019-03-12:
I have same issue, octoprint missing a lot of data when connected to duet wifi....can we reopen this?  What logs do you need?

foosel, 2019-03-13:
A `serial.log` from connecting to the printer, and an `octoprint.log`. Possibly more after a first analysis of both.

trondvedoy, 2019-08-09:
Hi, I got the same issue trying to connect a Duet2Wifi to my RPi running octoprint. 
The Duet has serial port **/dev/ttyACM0** - and I suspect baudrate is **9600** as I have been using that successfully between PC&Duet.
Serial.log & octoprint.log in the following link: https://1drv.ms/u/s!Akm99K6I5YFdrjAG15qsqF2mtAgz?e=rxYWtK

I will gladly assist if you need more logs or info.

jwoolston, 2019-08-12:
@trondvedoy I have been using a duet wifi through OctoPrint for months. The baud rate is 115200 but I don't bother setting it, the auto detect always works fine.

trondvedoy, 2019-08-24:
@jwoolston I never get it to connect, not even with auto detect. I have also tried all the settings manually.