OctoPrint/OctoPrint #3008: Traceback sarge.run

donnydavis, 2019-01-20:
latest version of octoprint / latest updates on octopi

logs show nothing except runs prior to the update. 
manual start of the octopi server yields the following traceback

Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 11, in <module>
    load_entry_point('OctoPrint==1.3.10', 'console_scripts', 'octoprint')()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/__init__.py", line 599, in main
    patch_sarge_async_on_py2()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 30, in patch_sarge_async_on_py2
    sarge.run = move_async(sarge.run)
AttributeError: 'module' object has no attribute 'run'

donnydavis, 2019-01-20:
It would appear that the sarge folder contains nothing in the text files, they are all empty. 

 pip install -I sarge
Collecting sarge
Exception:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 809, in unpack_url
    unpack_file_url(link, location, download_dir, hashes=hashes)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 715, in unpack_file_url
    unpack_file(from_path, location, content_type, link)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 599, in unpack_file
    flatten=not filename.endswith('.whl')
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 484, in unzip_file
    zip = zipfile.ZipFile(zipfp, allowZip64=True)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 813, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
BadZipfile: File is not a zip file

tedder, 2019-01-20:
Not sure about the zip file error, but your version of sarge is out of date. There was a breaking change in there.

closed 2019-02-04 by GitIssueBot