OctoPrint/OctoPrint #3009: Webcam feed extremely unreliable

Zylvian, 2019-01-20:
Two gifs explaining the situation: 
https://gyazo.com/c1e06126b0f2425b6859725b359531f8
https://gyazo.com/21ff2d4680bde4d239143cd20b9ba097

It seems to only sometimes actually stream, and when it does, it stops after a few seconds. Never seen this behavior before.

Is this a known issue that I couldn't find? Or are there any logical possible errors?

FanDjango, 2019-01-20:
I  also had/have problems with mjpg-streamer. My config: Logitech C270 camera, Tinkerboard S, Armbian/Debian.

Make sure you got it from this guy and check the issues here first: 
[https://github.com/jacksonliam/mjpg-streamer/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc](https://github.com/jacksonliam/mjpg-streamer/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc)

Also, for a test to see what is happening, ssh into your octoprint system and stop the webcam streamer daemon, start it manually as follows, observe the output when it fails to see if you can get a hint.
```
root@octo-a8:~# service webcam stop
root@octo-a8:~# cd /home/pi/scripts
root@octo-a8:/home/pi/scripts# ls
webcam  webcamDaemon
root@octo-a8:/home/pi/scripts# cat webcamDaemon
#!/bin/bash

#raspicam on asus tinkerboard
#MJPGSTREAMER_HOME=/home/pi/code/mjpg-streamer
#camera_options="-n -y -f 10"

#logitech c270
#MJPGSTREAMER_HOME=/home/pi/mjpg-streamer/mjpg-streamer-experimental
#camera_options="-n -r 1280x720 -f 30"

# runs MJPG Streamer, using the provided input plugin + configuration
function runMjpgStreamer {
    input=$1
    pushd /home/pi/mjpg-streamer/mjpg-streamer-experimental
    export LD_LIBRARY_PATH=/home/pi/mjpg-streamer/mjpg-streamer-experimental
    #./mjpg_streamer -o 'output_http.so -w ./www' -i 'input_uvc.so -r 1280x720 -f 30 -n -q 100'
    ./mjpg_streamer -o 'output_http.so -w ./www' -i 'input_uvc.so -n -r 1280x720 -f 30'
    popd
}

# starts up the RasPiCam
function startCam {
    runMjpgStreamer
}

# keep mjpg streamer running if some camera is attached
while true; do
    startCam
    sleep 5
    killall mjpg_streamer
    sleep 5
done

root@octo-a8:/home/pi/scripts# ../mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer -o 'output_http.so -w ./www' -i 'input_uvc.so -r 1280x720 -f 30'
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
```
Note: I did this as root. Not neccessary.

Another way ist to do **cat /var/log/syslog | grep streamer** and see if you get any error messages.

I am currently thinking about creating a cron task, that will pull a still image from the cam using mjpg-streamer, and if that fails or if a diff to the previous still image is equal, to stop and restart the webcam daemon. 

I am also getting pissed off by how often it just stops. Sometimes I am lucky and it runs for hours, sometimes it fails 5 times in a couple of minutes.

Zylvian, 2019-01-21:
https://i.gyazo.com/93295cdb374ad8014c9ccd0049482f54.mp4

Doesn't seem to be any errors at all after startup, it's just super unreliable.