OctoPrint/OctoPrint #3011: octoprint loses connection with a buffer overflow

adcurtin, 2019-01-23:
#### What were you doing?

1. start a print from octoprint

2. trigger p stop on the printer (e.g. run out of filament)

3. wait an hour, with the printer sending:

`Recv: // echo: Waiting for unpause button on the LCD panel
Recv: // echo: Waiting for unpause button on the LCD panel
Recv:  T:227 /230 B:41 /0 T0:227 /230 T1:42 /0 @:0 B@:0
Recv: // echo: Waiting for unpause button on the LCD panel
Recv: // echo: Waiting for unpause button on the LCD panel
Recv:  T:227 /230 B:40 /0 T0:227 /230 T1:42 /0 @:0 B@:0`
 over and over

4. buffer overflow!

`Unexpected error while reading serial port, please consult octoprint.log for details: BufferOverflow: 'Buffer overflow' @ comm.py:_readline:2605
Changing monitoring state from "Printing" to "Offline (Error: BufferOverflow: 'Buffer overflow' @ comm.py:_readline:2605)"
Connection closed, closing down monitor
Closing down send loop`

#### What did you expect to happen?

I expected octoprint to handle this pause, or at least just sit there and wait until I got there to fix the printer

#### What happened instead?

octoprint disconnected after an hour

#### Did the same happen when running OctoPrint in safe mode?

No. GPX can't run in safe mode, so I can't connect to the printer.

#### Version of OctoPrint

Version 1.3.10

#### Operating System running OctoPrint

octopi  Version 0.14.0

#### Printer model & used firmware incl. version

powerspec 3d pro with sailfish 7.7

#### Browser and version of browser, operating system running browser

n/a

#### Link to octoprint.log

https://pastebin.com/9Y8QEzs0

#### Link to contents of terminal tab or serial.log

https://pastebin.com/rAqaCgwx

I have serial.log, but don't think more than this is necessary.

#### Link to contents of Javascript console in the browser

n/a

#### Screenshot(s)/video(s) showing the problem:

n/a

I have read the FAQ.

DanielJoyce, 2019-02-05:
Your octoprint log shows these errors:

2019-01-22 21:32:48,684 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Printing"
2019-01-22 21:32:48,692 - octoprint.plugin - ERROR - Error while calling plugin tracking
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 230, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking/__init__.py", line 119, in on_event
    self._track_printjob_event(event, payload)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking/__init__.py", line 237, in _track_printjob_event
    sha.update(self._settings.get([b"unique_id"]))
TypeError: must be string or buffer, not None

I dunno if it might play a role, but thats a bit weird one too.

adcurtin, 2019-02-05:
there's a good chance this could be a problem in the GPX plugin instead of octoprint. If signs are pointing toward that, I'd be happy to go report this bug there.

foosel, 2019-02-08:
> I dunno if it might play a role, but thats a bit weird one too.

That's a bug in the tracking plugin that's been fixed for 1.3.11 already and unrelated.

@adcurtin Actually, my money would be very much on this being a GPX issue here. I'd prefer if this was ruled out first.

foosel, 2019-10-31:
Closing this since it's tracked on the GPX plugin repo. Should that at some point result in this being an OctoPrint bug I'll happily reopen this.

closed 2019-10-31 by foosel