OctoPrint/OctoPrint #3012: error messages after installing plugin, KO language pack

wooooooooooook, 2019-01-25:
#### What were you doing?

1. korean language pack has been installed - https://github.com/foosel/OctoPrint/wiki#language-packs
2. Install any plugin
3. Plugin installed successfully but error message
4. restart octoprint --> plugin works well
5. just error message

#### What did you expect to happen?
Install plugin without error message

#### What happened instead?
Install plugin successfully but error message

#### Did the same happen when running OctoPrint in safe mode?
Installing plugin is not available in safe mode

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
octopi 0.15.1

#### Printer model & used firmware incl. version
ender-2, klipper

#### Browser and version of browser, operating system running browser
chrome latest, windows 10

#### Link to octoprint.log
https://pastebin.com/XXrKnVQa

#### Link to contents of terminal tab or serial.log
nothing happens in terminal tab

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

I think it is encoding issue because it says :
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-1: ordinal not in range(128)

Is this a translation error?

foosel, 2019-01-25:
No, that looks like a bug in OctoPrint. The plugin probably translates the name of one of the core plugins to korean, then when trying to print the list of installed plugins to the log file this error is encountered because the name no longer fits into ASCII. Not sure why I even chose `str` there to begin with, it really should be `unicode`. Will look into a fix for 1.3.11.

foosel, 2019-01-25:
Should be fixed by the above commit. Pushed to `maintenance`, soon `devel`, to be included in 1.3.11.

From the looks of it, the translation of "Backup & Restore" in the language pack (plus that name even being translatable on system level) are the problem.

Until 1.3.11 is released I suggest to use a language pack that doesn't translate "Backup & Restore". That should work around the issue. 

This adjusted Korean one worked at least in my test: [OctoPrint-i18n-ko_20190125150900.zip](https://github.com/foosel/OctoPrint/files/2796647/OctoPrint-i18n-ko_20190125150900.zip)

wooooooooooook, 2019-01-25:
> Should be fixed by the above commit. Pushed to `maintenance`, soon `devel`, to be included in 1.3.11.
>
> From the looks of it, the translation of "Backup & Restore" in the language pack (plus that [...]

Thanks a lot! 
It works well

commit 09d3c441 (2019-01-25): :bug: unicode-ify octoprint.plugin.core

Fixes #3012
closed 2019-05-14 by foosel