OctoPrint/OctoPrint #3014: UnpicklingError: invalid load key

tikaani, 2019-01-26:
Running OctoPrint on a raspberry which keeps giving this error after reboot.

--- Logging ---
2019-01-26 14:56:55,309 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 972234752
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Rev 1.2
|  python:
|    pip: '18.1'
|    version: 2.7.13
|    virtualenv: /home/pi/OctoPrint/venv
2019-01-26 14:56:55,318 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2019-01-26 14:56:55,320 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2019-01-26 14:56:55,650 - octoprint.server - INFO - Shutting down intermediary server...
2019-01-26 14:56:55,989 - octoprint.server - INFO - Intermediary server shut down
2019-01-26 14:56:55,995 - octoprint.events - INFO - Processing startup event, this is our first event
2019-01-26 14:56:55,996 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2019-01-26 14:56:56,000 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2019-01-26 14:56:56,014 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2019-01-26 14:56:56,161 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2019-01-26 14:56:56,176 - octoprint.plugins.cura - INFO - Path to CuraEngine has not been configured or does not exist (currently set to None), Cura will not be selectable for slicing
2019-01-26 14:56:56,190 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2019-01-26 14:56:56,197 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on Silver3D for SSDP
2019-01-26 14:56:56,249 - octoprint.server - INFO - Listening on http://[::]:5000
2019-01-26 14:56:56,283 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2019-01-26 14:56:57,191 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2019-01-26 14:56:58,033 - octoprint.util.pip - INFO - Using "/home/pi/OctoPrint/venv/bin/python2 -m pip" as command to invoke pip
2019-01-26 14:56:58,035 - octoprint.util.pip - INFO - pip installs to /home/pi/OctoPrint/venv/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-01-26 14:56:58,037 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-01-26 14:56:58,077 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2019-01-26 14:56:58,170 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2019-01-26 14:56:58,946 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2019-01-26 14:56:59,041 - octoprint.plugins.tracking - INFO - Sent tracking event startup, payload: {'python': '2.7.13', 'ram': 972234752, 'version': u'1.3.10', 'pip': '18.1', 'cores': 4, 'freq': 1200.0, 'pi_model': 'Raspberry Pi 3 Model B Rev 1.2', 'os': 'linux'}
2019-01-26 15:03:21,585 - octoprint.server.util.flask - INFO - Passively logging in user SilverOcto from 10.0.1.202
2019-01-26 15:03:21,708 - octoprint - ERROR - Exception on / [GET]
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/views.py", line 412, in index
    response = default_view()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/views.py", line 355, in default_view
    now)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/views.py", line 423, in _get_render_kwargs
    locales = dict((l.language, dict(language=l.language, display=l.display_name, english=l.english_name)) for l in LOCALES)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/views.py", line 423, in <genexpr>
    locales = dict((l.language, dict(language=l.language, display=l.display_name, english=l.english_name)) for l in LOCALES)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/babel/core.py", line 381, in get_display_name
    retval = locale.languages.get(self.language)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/babel/core.py", line 482, in languages
    return self._data['languages']
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/babel/core.py", line 364, in _data
    self.__data = localedata.LocaleDataDict(localedata.load(str(self)))
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/babel/localedata.py", line 115, in load
    merge(data, pickle.load(fileobj))
UnpicklingError: invalid load key, ' '.
2019-01-26 15:03:21,712 - tornado.access - ERROR - 500 GET / (10.0.1.202) 680.39ms

foosel, 2019-01-28:
Do you have any additional language packs installed?

tikaani, 2019-01-31:
I installed OctoPrint maually after I installed the base raspberry Raspbian Lite Image.
with the installation of this image I did not installed any extra language packs, so it defaults to en_GB.UTF-8.
Btw. this error occurred after a reboot while it was working before

foosel, 2019-02-07:
What preceded that reboot? Did you shut down/reboot the Pi properly through the system menu or did you just pull power?

tikaani, 2019-02-07:
I shutdown the Pi via the command terminal with sudo shutdown

foosel, 2019-02-07:
Then I have no idea what happened here. It's hiccuping on what looks to be corrupted language data, and since you say you haven't installed and additional OctoPrint language packs it must be a bundled language pack that got corrupted in your install. I can mitigate the error in 1.3.11 (and in fact already have as of 3815c340a87d42c45223d62b75676c26e1c24da0), but how it got corrupted in the first place I cannot tell you, never see that happening before. I suggest to reinstall & restart:

```
source ~/oprint/bin/activate
pip install --force-reinstall https://get.octoprint.org/latest
sudo service octoprint restart
```

Marking as solved since the symptom has been fixed on `maintenance` and `devel` and said fix will be part of 1.3.11.

PS: Next time please fill out the full ticket template.

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel