OctoPrint/OctoPrint #3015: 'Cancel Job' waiting for the bed heated.

halxinate, 2019-01-26:
Version 1.3.10

When clicking 'Cancel' at the very beginning of the print job it strays in the 'canceling' state until the bed temp is reached. On my printer, it takes several minutes which is a rude waste of my time. Please, find a solution.

JustAnother1, 2019-01-28:
I had a similar issue, and a possible solution was probably suggested before.
As I understand it the issue is the M190 command. It is defined to wait until the bed reaches the commanded temperature. You need to use them in gcode files as you want the printer to wait.

For Octoprint using this command creates a "long running" command. The printer _should_ not read the next command before finishing this one. So there might be work around in some firmware that help, but it really is the way it is supposed to be.

The possible solution would be for Octoprint to detect that the command is a M190 and to replace it with a M140. The M140 just configures the heater and returns immediately. Octoprint could then monitor the heated bed temperature until it reaches the level defined in the M190 command. Once the bed is heated Octoprint could then continue sending commands. This would allow Octoprint to talk to the printer during the "long running" command.

foosel, 2019-01-28:
There's [FAQ entry on this](https://discourse.octoprint.org/t/octoprint-wont-cancel-when-my-printer-is-heating-up/229). So this ticket is a rude waste of *my* time.

closed 2019-01-28 by foosel