OctoPrint/OctoPrint #3022: Upload folder not being read

Monkeyattack, 2019-01-29:
All of a sudden, I can not upload gcode or stl files successfully.  The upload runs and hangs at "Saving".  

The octoprint.log looks successful (full log linked below).

`2019-01-29 17:28:20,316 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/Workspace_2019-01-25_12_22_05.gcode
2019-01-29 17:29:20,732 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Workspace_2019-01-25_12_22_05.gcode finished, needed 60.42s
`

[octoprint.log](https://gist.github.com/Monkeyattack/19fc670336e3b15c83a7a1b93424bf2a)

After a failed upload the file listing section of the page no longer returns the previously listed files nor does it even display space information.

Screenshot:
https://i.imgur.com/v5mbv7l.png

I have tried running in Safe Mode, but it doesn't make a difference.

If I SSH in and delete everything from the uploads folder, and delete the .metadata.json I can sometimes get the next file to work.  The regenerated metadeta file has all the info for the files that are no longer there still though,

[Console and .metadata.json file](https://gist.github.com/Monkeyattack/accb45b359a4fbd59435847ccdf78377)

Screenshot after:
https://i.imgur.com/6S43VQp.png

OS:
Linux octopi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l

Version:
OctoPrint version : 1.3.10
OctoPi version    : 0.15.1

Raspberry Pi 3B

Printer: Monoprice Ultimate

Marlin firmware

foosel, 2019-01-30:
I agree, from the log it looks like the upload finished, otherwise it wouldn't have anything to analyse.

I've never seen something like this so far and hence am currently a bit at a loss. Things to try/provide:

  * Please take a look into the Network tab of the browser developer tools and see what state the upload request is in. It should be `POST` request to `/api/files/local`.
  * What happens if you shutdown OctoPrint, delete/rename the metadata file, then restart OctoPrint and try to upload a file?
  * Does the start of this issue correlate with any changes to your setup of any kind? Incomplete shutdown, new plugin installed, update of OctoPrint, anything like that?

I'll also grab your metadata file (thanks for providing that!) and try to replicate the behaviour.

Monkeyattack, 2019-01-30:
I see the post
[Network Tab of Browser](https://i.imgur.com/6DD4iij.png)

Attached log after another upload
[octoprint.log](https://gist.github.com/Monkeyattack/05cfe8e8fe7e2d5957cf096a01dbfafe)

I added a 3.5 inch touchscreen a week before the problem or so and installed TouchUI.  I tried removing it, and even installed octopi fresh on a new sd card and without the touchscreen.  It started doing the same thing, so I switch back to my original install.  I have removed the touchscreen, since the case I was using cause occasional temperature errors.

If I delete the upload folder contents and upload with a new filename it works.  If I try with the same filename that is already in the .metadata.json it will hang and not work.

I'll try swapping out my PI and the OS.  Maybe there is a physical problem.

Salandora, 2019-01-30:
This looks like a problem in the browser to me, also the little picture you attached states there are 2 errors,
Can you show the output of the Console Tab please.
Also expand the error lines as much as possible.

foosel, 2019-01-31:
@Monkeyattack your pi is overheating:

```
2019-01-29 16:52:26,495 - octoprint.plugins.pi_support - WARNING - This Raspberry Pi is reporting problems that might lead to bad performance or errors caused by overheating or insufficient power.
!!! FREQUENCY CAPPING DUE TO OVERHEATING REPORTED !!! Improve cooling on the Pi's CPU and GPU.
```

This should be looked into and solved before we continue here. Throttled Pis (due to overheat or undervoltage situations) are known to develop the strangest behaviours, and I want to rule out that this is the contributing factor here.

> EDIT:
> have you tried to start octoprint in safe mode? Could also be a plugin messing something up I guess.

According to the filled out ticket template above, they did.

psynautic, 2019-02-18:
I have this same problem. I solved it by uninstalling, removing all remnants of filamentmanager plugin. and then deleting the metadata file.  :[

maelremrem, 2019-03-06:
Same problem here. I have tried on a Mac and on Windows with differents browser. I can print with the astroprint plugin. I upload to astroprint and then I push the file from the astroprint cloud to my octoprint instance and it work. But still no file shown. I have also looked in the folder (via SSH) and the files are inside. I have also tried to create a folder, it's not shown on octoprint but it's actually on the pi (looked via SSH)

foosel, 2019-03-08:
From the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report):

> ### What should I include in a bug report?
> 
> First of all make sure your use a descriptive title. "It doesn't work" and similar unspecific complaints are > NOT descriptive titles.
> 
> **Always use the following template, even if only adding a "me too" to an existing ticket:**
>
> [...]

psynautic, 2019-03-10:
I think I made a discovery here.  My files api request and my .metadata.json file has `Infinity` in `EstimatedPrintTime`. I don't think `Infinity` is a json spec value. If i fix this to just have an arbitrary float, all the files load!

[my files response](https://gist.github.com/psynautic/277b5f1e3da718752e716219428f7f7d)

kazooless, 2019-03-13:
I experienced the same problem and assumed I had an SD Card going bad so I spent the night installing from scratch (I have a Cubox-i so I have to install Debian and configure everything manually). Anyway, I had the same problem described above but when I read the last comment I took a look. my metadata.json file also had an EstimatedPrintTime without a number (it had NaN) but changing it didn't help. However, it was referring to a particular file I got from somewhere besides Cura. As soon as I deleted the gcode file, the directory refreshed just fine.

The file that was causing this was generated by the Marlin Firmware site for calibrating the K-Factor for Linear Advance.

foosel, 2019-03-13:
@kazooless could you share that file?

shvbox, 2019-03-29:
The same issue. Occurs when G-code contains "NaN" instead of numbers. Just delete the problem file in uploads folder on server (via ftp or ssh). Then update the file list in the Octoprint interface.

alxpx, 2019-03-30:
This is a "Me, too" for the issue expressed by Monkeyattack.  The behavior and circumstances very closely match those expressed by kazooless, as well (uploading Linear Advance K-Factor calibration Gcode generated by the page [http://marlinfw.org/tools/lin_advance/k-factor.html](http://marlinfw.org/tools/lin_advance/k-factor.html).

I have already flashed a new image on my pi, so I'm presently unable to test the steps mentioned by shvbox, though I will repro and document if I run into the issue again.

#### What were you doing?

Uploading GCode generated at [http://marlinfw.org/tools/lin_advance/k-factor.html](http://marlinfw.org/tools/lin_advance/k-factor.html) for calibrating the Linear Advance K-Factor in Marlin.

1. Initially, the behavior presented as a hanging upload.  The progress bar went all the way to the end, but the upload would not complete.  After rebooting the Pi, I was able to successfully upload the file on my second attempt.

2. After the upload, I tried to launch a print for the file on another device, but the print failed to start.

3. Thinking it was a UI issue on the device (iPad running Safari), I reloaded the page, after which, the "Files" frame was completely empty.

4. I rebooted the Pi via the browser and PC I'd been using initially, but had the same behavior.

5. To confirm the SD card wasn't damaged, I accessed the folder via SSH and was able to view all files in the console.

6. Attempts to refresh the Files frame did not seem to process, nor did changing the sorting method, though the "Create Folder" dialogue did successfully create a new folder on the SD, though the new folder was only visible via the shell session.

7. Rebooting in Safe Mode exhibited the same behavior.

8. Setting the mode bits for the ~/.octoprint/uploads folder and a test file to allow RWX privledges for all users/groups did not change the behavior.

9. Behavior was confirmed on two devices and three browsers: Windows 10 x64 (Edge and Opera), and iPad (Safari).

10. Updating from OctoPrint 1.3.10 on OctoPi 0.16.0 to OctoPrint 1.3.11rc1 and rebooting in Safe Mode did not resolve the behavior.

#### What did you expect to happen?

Files to be queried and displayed within the Files frame.

#### What happened instead?

Uploads cannot complete, and although the query "GET /api/files?recursive=true&force=true" appears to be pulling information on the files and folders successfully, results of the query aren't being populated within the Files frame.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.3.10 and OctoPrint 1.3.11rc1 running on OctoPi 0.16.0

#### Operating System running OctoPrint

Raspbian (OctoPi 0.16.0)

#### Printer model & used firmware incl. version

Creality CR-10, MKS Gen L motherboard, Marlin 1.1.9

#### Browser and version of browser, operating system running browser

Opera 64-bit 58.0.3135.118, Microsoft Edge 42.17134.1.0: Windows 10 Pro 64-bit

Safari Mobile 9.0 on iOS 9.3.5

#### Link to octoprint.log

https://gist.githubusercontent.com/alxpx/835a5c3c7fa398ed54fee3972fd1cd36/raw/142e417b6ba07a03e4b749da64a5c708192b2de5/octoprint.log

#### Link to contents of terminal tab or serial.log

https://gist.github.com/alxpx/4dad0ceef5edb39ee86595dbb1fcacfe

#### Link to contents of Javascript console in the browser

https://gist.github.com/alxpx/0d59bf434633af46a7f28d822265a381

#### Screenshot(s)/video(s) showing the problem:

http://i.imgur.com/58FABFg.png

I have read the FAQ.

foosel, 2019-05-08:
So the problem here is that the GCODE literally contains `ZNaN` which doesn't make a whole lot of sense and I doubt that would make any firmware that receives that very happy:

```
G1 X162.5 Y98 ENaN F1200 ; 0
G1 X162.5 Y100 ENaN F1200 ; 0
G1 X162.5 Y102 ENaN F1200 ; 0
G1 X160.5 Y102 ENaN F1200 ; 0
G1 X160.5 Y100 ENaN F1200 ; 0
G1 X160.5 Y98 ENaN F1200 ; 0
```

```
G1 ZNaN F900 ; zHop
```

The GCODE analyser is so far capable of parsing this however since `NaN` is a valid Python float string. I'll harden the analyser so that it ignores any coordinates or other values in GCODE that are `NaN` or `Inf` and will also take a look at automatically cleaning up metadata that's broken in that way on initial start. 

I'm currently not able to generate such broken lines with the linked tool, however it might be some specific combination of parameters that achieves that in which case it should probably be reported to the Marlin project.

foosel, 2019-05-08:
Should be solved by the above commits, both on `maintenance`, soon `devel`, to be part of 1.3.12.

psynautic, 2019-05-08:
you are a saint, this was a really weird issue!

thinkyhead, 2019-05-09:
@Sineos — It looks like `k-factor.js` is acting up sometimes. (Maybe the form wasn't fully populated?) I could try to patch the `k-factor.js` script to return `0` instead of `NaN`, but that wouldn't fix the core issue. I'm sure you can quickly figure out the cause of these `NaN`s falling into the G-code output. You can just point me towards a fix or post a PR. (You may need to `git fetch upstream ; git reset --hard upstream/master` to get the latest `MarlinFirmware/MarlinDocumentation` content into your working copy.)

Sineos, 2019-05-09:
@thinkyhead I will check and post a PR

commit b79fc94a (2019-05-08): ☔ Protect against NaN & Inf in GCODE

Also cleanup broken metadata entries that contain invalid json.

Fixes #3022
closed 2019-10-22 by foosel