OctoPrint/OctoPrint #3023: string indices must be integers, not unicode during installation

dedalo88, 2019-01-30:
#### What were you doing?

I started the Octoprint setup following this instructions page:
https://discourse.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337

```
cd ~
sudo apt update
sudo apt install python-pip python-dev python-setuptools python-virtualenv git libyaml-dev build-essential
mkdir OctoPrint && cd OctoPrint
virtualenv venv
source venv/bin/activate
pip install pip --upgrade
pip install https://get.octoprint.org/latest
sudo usermod -a -G tty myUser
sudo usermod -a -G dialout myUser
```
The first time 2018-09-14 the setup worked well, and I was able to reach http://127.0.0.1:5000
Then on 2019-01-27 I updated the system with apt-get and Octoprint following setup again, but I get http error 500:
```
2019-01-27 17:50:12,049 - octoprint.startup - INFO - ******************************************************************************
2019-01-27 17:50:12,053 - octoprint.startup - INFO - Starting OctoPrint 1.3.9
2019-01-27 17:50:12,054 - octoprint.startup - INFO - ******************************************************************************
2019-01-27 17:50:12,167 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins, /home/technology/.octoprint/plugins and installed plugin packages...
2019-01-27 17:50:13,893 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2019-01-27 17:50:14,671 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2019-01-27 17:50:14,683 - octoprint.plugin.core - INFO - Found 10 plugin(s) providing 10 mixin implementations, 8 hook handlers
2019-01-27 17:50:14,816 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-01-27 17:50:14,895 - octoprint.server - INFO - Intermediary server started
2019-01-27 17:50:14,896 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins, /home/technology/.octoprint/plugins and installed plugin packages...
2019-01-27 17:50:14,911 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2019-01-27 17:50:15,489 - octoprint.plugin.core - INFO - Found 10 plugin(s) providing 10 mixin implementations, 8 hook handlers
2019-01-27 17:50:15,542 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/technology/.octoprint/uploads...
2019-01-27 17:50:15,545 - octoprint.filemanager.storage - INFO - ... file metadata for /home/technology/.octoprint/uploads initialized successfully.
2019-01-27 17:50:15,592 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2019-01-27 17:50:21,437 - octoprint.util.pip - INFO - Using "/usr/bin/python -m pip" as command to invoke pip
2019-01-27 17:50:26,846 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2019-01-27 17:50:26,850 - octoprint.util.pip - INFO - pip installs to /usr/lib/python2.7/dist-packages (writable -> no), --user flag needed -> yes, virtual env -> no
2019-01-27 17:50:26,851 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-01-27 17:50:26,856 - octoprint.plugin.core - INFO - Initialized 10 plugin implementation(s)
2019-01-27 17:50:26,866 - octoprint.plugin.core - INFO - 10 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/discovery
|  Logging (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/logging
|  Plugin Manager (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/printer_safety_check
|  Software Update (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/virtual_printer
2019-01-27 17:50:26,886 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 484048896
|  os:
|    id: linux
|    platform: linux2
|  python:
|    pip: 9.0.1
|    version: 2.7.13
2019-01-27 17:50:26,900 - octoprint.server - INFO - Reset webasset folder /home/technology/.octoprint/generated/webassets...
2019-01-27 17:50:26,905 - octoprint.server - INFO - Reset webasset folder /home/technology/.octoprint/generated/.webassets-cache...
2019-01-27 17:50:27,255 - octoprint.server - INFO - Shutting down intermediary server...
2019-01-27 17:50:27,427 - octoprint.server - INFO - Intermediary server shut down
2019-01-27 17:50:27,431 - octoprint.events - INFO - Processing startup event, this is our first event
2019-01-27 17:50:27,432 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2019-01-27 17:50:27,435 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2019-01-27 17:50:27,442 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
[... 151 lines cut ...]
#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.3.9

#### Operating System running OctoPrint
ARMBIAN 5.70 stable Debian GNU/Linux 9 (stretch) 4.19.13-sunxi

#### Printer model & used firmware incl. version
na

#### Browser and version of browser, operating system running browser
Firefox, Windows 10

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.
[octoprint.log](https://github.com/foosel/OctoPrint/files/2812117/octoprint.log)

foosel, 2019-01-30:
Looks like for some reason there's some inconsistent internal state in the preemptive cache that leads to misformed data. I don't know yet why or how that's possible and will have to take a look. 

To get things working for you again, please try disabling preemptive caching by setting `devel.cache.preemptive` in `config.yaml` to `false`. See also [the docs here](http://docs.octoprint.org/en/master/configuration/config_yaml.html#development-settings).

Side note: It looks like you have OctoPrint installed globally. That is discouraged, the supported way is to use a virtual environment to rule out any system package versions from causing problems (*might* be a factor here, but only *might*). Also 1.3.10 has been out since early December.

dedalo88, 2019-01-31:
I updated the `config.yaml` file as you indicate and it worked. 
Actually I don't know the difference between global installation and virtual environment installation, where could I make things wrong?
This is my `preemptive_cache_config.yaml` file:

```
V//
```

dedalo88, 2019-02-01:
yes I did, I installed octoprint until the web wizard start, then I turned the board off to continue later.
Thank you for support :)

commit 1aea92a5 (2019-01-30): ☔ Protect against invalid preemptive cache config data

Fixes #3023
closed 2019-05-14 by foosel