OctoPrint/OctoPrint #3024: Inconsistent behaviour of CORS support on API vs static routes vs socket

foosel, 2019-01-31:
### Problem

  1) If CORS access is enabled, the static Tornado endpoints (such as downloads) don't allow access
  2) If CORS access is disabled, SockJS still allows CORS access

See also [this discussion on the community forum](https://discourse.octoprint.org/t/cors-error-when-fetching-gcode-file-from-another-web-application/).

### Solution

Two fold: 

  1) Add CORS headers to the tornado endpoints if it's enabled
  2) Remove CORS headers from SockJS endpoints if it's disabled

fabian-h, 2019-01-31:
One additional aspect I discovered while looking into this was that web sockets are by design not covered by CORS (see e.g. https://blog.securityevaluators.com/websockets-not-bound-by-cors-does-this-mean-2e7819374acc.). 

So as far as I understand it, setting the CORS headers won't do anything there, a similar mechanism has to be explicitly coded into the web socket server that checks the origin and allows or disallows the connection. Not sure if SockJS already has some mechanism like this.

foosel, 2019-01-31:
It actually has, I'm currently looking at that. It only has that on the Websocket transport though (obviously), the fallback transports and handshake endpoints are covered by regular CORS and are currently hard coded to allow everything. Trying to figure out how that can be solved. The websocket behaviour can be controlled through a setting at least.

foosel, 2019-02-01:
Should be fixed on `maintenance` and `devel` per the above commits and will be part of 1.3.11.

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel