OctoPrint/OctoPrint #3028: Cannot update Octoprint

noisemaker00, 2019-02-01:
Using the update plugin always return an error, which report:
```
Could not fetch URL https://pypi.python.org/simple/sarge/: There was a problem confirming the ssl certificate: ("bad handshake: Error([('SSL routines', 'SSL3_GET_SERVER_CERTIFICATE', 'certificate verify failed')],)",) - skipping
Could not find a version that satisfies the requirement sarge==0.1.5post0 (from OctoPrint==1.3.10) (from versions: )
No matching distribution found for sarge==0.1.5post0 (from OctoPrint==1.3.10)
You are using pip version 8.1.1, however version 18.0 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

tedder, 2019-02-01:
What version of Octoprint are you on?

What happens if you `curl https://pypi.python.org/simple/sarge/`?

galamdring, 2019-02-08:
When I run pip search sarge, that is the version that comes up. I would try updating pip, as the version you are running is excessively old, and might be the problem.

closed 2019-02-16 by GitIssueBot