OctoPrint/OctoPrint #3033: [Request] Have OctoPrint open Serial port in Exclusive access mode on platforms that support it.

DanielJoyce, 2019-02-04:
**Is your feature request related to a problem? Please describe.**

Reading some online documents about Octoprint serial connectivity issues, it seems there is a reccomendation of making sure no other process is using the port. PySerial since 3.0 supports opening serial ports in exclusive access mode on POSIX systems.

This would prevent services from trying to use the same port, and also make debugging problems easier as OctoPrint would get an error when trying to open a port already open by some other process.

**Describe the solution you'd like**

Octoprint should by default open serial ports in exclusive access mode

https://pyserial.readthedocs.io/en/latest/pyserial_api.html

**Describe alternatives you've considered**
There really aren't any.

DanielJoyce, 2019-02-08:
Merged so this should be marked solved too.

foosel, 2019-02-09:
Odd, I thought I did

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel