OctoPrint/OctoPrint #3037: [Request] Webpage print progressbar larger 100%

njozsef, 2019-02-06:
progressbar larger than 100.
screenshot: https://photos.app.goo.gl/9pfjaNfdwHmUaJ2L7
objet location: https://www.thingiverse.com/thing:2018624

version:
OctoPrint 1.3.10 running on OctoPi 0.15.1

Thank you
Jozsef

foosel, 2019-02-07:
This looks less like a feature request (which you marked it as) and more like a big report. And to look into that I'll need way more info to be able to look into it. Please fill out a full ticket template.

njozsef, 2019-02-07:
Thank you. Yes this is not req this is info, sorry :)

njozsef, 2019-02-07:
Hi Gina! and Hi GitIssueBot!
I tested this problem safe mode.
This you software is great and working good!
Sorry for disturbing.

Maybe 3rd progressbar plugin is wrong.
Only if pressed 2nd  or more printing same (loaded) object.
video: 
first print w plugins (good percent): https://youtu.be/jVWZY6qowEc
secondary print with plugins (bad percent): https://youtu.be/jVWZY6qowEc?t=263

safe mode is all good no problem: https://youtu.be/jVWZY6qowEc?t=584

plugins:
Detailed Progress Plugin (0.1.4)
PrintTimeGenius Plugin (1.3)

Thank you the fast response
best regards Jozsef
Please close

closed 2019-02-07 by njozsef