OctoPrint/OctoPrint #3040: Print time estimates rounding to hours

jonnieZG, 2019-02-07:
As of 1.3.9, hooks for adding custom, more accurate time estimators have been added. However, when OctoPrint receives these estimates, it still rounds them to hours when remaining time is >1h, then to 10 minutes when time is between 1 hour and 30 mins, and only when the remaining time is below 30 mins it starts showing actual times returned by a plugin.

**Current Behavior**

I am using https://github.com/sysadminsh/OctoPrint-M73ETAOverride (in fact, my own branch https://github.com/jonnieZG/OctoPrint-M73ETAOverride) that returns estimates generated by Slic3r, that are *very* accurate and precise to 1 min, but instead of showing "Print time left: 2 hours 32 minutes", it shows "3 hours", "2 hours", "1 hour", "50 minutes", "40 minutes" ... "29 minutes", "28 minutes" etc.. which renders the whole plugin thing quite useless.

**Desired Behavior**

Times received by a time estimator plugin (that is currently supporting 1 sec resolution), should be displayed as "days:hours:minutes" without rounding just to the next bigger unit.

galamdring, 2019-02-08:
Is the estimation Octoprint is displaying wrong, or just a simplified view? It sounds like it is rounding to compress the information.

foosel, 2019-02-08:
The above commit adds a new setting you can switch off to disable time rounding/fuzziness (which only happens in the UI anyhow).

For the record, it doesn't round to the next unit, it rounds to the next half unit, with the unit depending on the duration:

https://github.com/foosel/OctoPrint/blob/61431c0df3fb571bd5fdaa6a910e4366a2145c51/src/octoprint/static/js/app/helpers.js#L437-L451

Personally I will probably never get why it is so important whether a print still will take 3h, 2h46min23s or 3h13min53s, it has the same impact on any kind of planning I need to do around the print, but whatever paddles all your collective canoes 🤷‍♀️

galamdring, 2019-02-08:
Beautiful! Thank you. I was looking to see how to do exactly that :)

jonnieZG, 2019-02-08:
Thank you, [foosel](/foosel)!

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel