OctoPrint/OctoPrint #3042: [Request] User configurable pause resume button

Hywelmartin, 2019-02-07:
I use Klipper and I use gCode macros to handle pause and resume so that I have a identical behavior if I use virtualSD-card or stream the gCode from Octoprint and make use of M600 inside the gcode
[image]

I would like to have the Pause button configurable so that when pressed it will send, in my case M600.
Likewise the "play"button but then I send RESUME

Today I have additional buttons and i the klipper plugin, they are on a different tab and small in the touch interface.

It would make the experience much more polished..

foosel, 2019-02-07:
Why don't you just configure custom pause and resume GCODE scripts to contain your macros?

Hywelmartin, 2019-02-07:
Well for starters.. I'm not a coder(I would have liked it thou) 
Second It doesen't work..
Octoprint does not recognize that the printer is paused and as I mentioned in the first post, I want identical behavior either printing thru Octoprint (local or SD) or if I start from the display..

foosel, 2019-02-08:
> Octoprint does not recognize that the printer is paused and as I mentioned in the first post, I want identical behavior either printing thru Octoprint (local or SD) or if I start from the display..

Then the firmware needs to tell OctoPrint that it is paused, overriding any kind of default buttons won't fix that. There are [mechanisms for that in place and supported by OctoPrint](http://docs.octoprint.org/en/master/features/action_commands.html).

> Well for starters.. I'm not a coder(I would have liked it thou)
> Second It doesen't work..

If you could explain your usage scenario a bit more it might be possible to help you make it work. Right now the only thing I know is a) you need to pause via the Klipper plugin (and I have no idea what that does, how it does it and so on) and b) pausing via the default buttons doesn't work in some unspecified way.

I understand Klipper basically replicates the locally stored files as if they were stored on the printer's SD here and is responsible of driving the print job? In that case I'd expect it to also react to the common SD commands for starting, pausing and resetting a print, which OctoPrint will send in case of an SD print when you use the default buttons, and also to send action commands as needed in case of external state manipulation. Apparently that is not the case here though so I need more information *how Klipper does things here* and also how the plugin works.

Swapping core functionality like what the print and pause buttons do is not a trivial change with a ton of possible side effects, so I really want to understand the problem space first. I need more information on the full Klipper workflow at play here and how it differs from a "regular" streamed or SD based print.

Hywelmartin, 2019-02-08:
> Octoprint does not recognize...

Maybe I was a little to fast here, there is a ongoing fix for the "fromdisplayprintfromsd" but there is a big job as I understand.. so that part is out of the Q. the normal "SD" print "works" as any other printer FW I assume.
And I even got aware of another ["fix"](https://github.com/Microtome/OctoPrint/tree/add-klipper-to-firmware-autodetect) for another problem..  That I didn't mentioned... some leftover moves that happened after the Pause due to a missing ACK (or what its called) due to human  readable "gcode"

` unknownCommandsNeedAck: true`
is needed

vSD-printing is as you describes  and [this](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#g-code-sd-card-commands) is the available gcodes

My use case is normally "vSD" printing to offload the RPi, I normally start the print within Octoprint that is merely a GUI to my printer. for simple things like jogging the axis, changing temps, fan speed and and so on..
Near my printer I use the touch GUI(or from small devices) otherwise the normal GUI.
for color changes I add M600 in the gcode, and for "unplanned" filament swap I use the macro buttons in the OctoKlipper plugin or call the command from the terminal in Octoprint.

Back to the request

Klippers built in Pause/resume function handles position velocity and E-steps internally so there is no "programming" needed("" gcode is some sort of programming"").

Klipper has built in capability for gcode macros. so everything can be handled "internally"..

For a simple pause the readable "gcode" is PAUSE, but that does just that (and take some sort of snapshot). and to resume the "gcode" is RESUME(and restore the snapshot).

So to call Klipper to pause a print from Octoprint there would be something else than just PAUSE because it would only render a blob and a melted part, and here could Octoprints "do after pause" scripts make it move somewhere else. But so does the macros in Klipper as well. 

would it be a solution that a plugin "hides" the button with there own? 

PRO
The convenience to have all settings in one place..
Ease of use for new users to the FW..
bridge Octoprints shortcomings regarding  information gathering during SD print and multiextruder configurations. documented [here](http://docs.octoprint.org/en/master/features/gcode_scripts.html)

CON
changes in the main code
a small user group

 /Martin

OutsourcedGuru, 2019-03-05:
Reading this, I'm fairly confused.

* It sounds like Klipper doesn't use `M0` as a pause control command. Instead, it wants to see `PAUSE`. Correct?
* Is it possible for you to first test some of these theories by adding custom `Pause` and `Resume` Control buttons in combination with the [Custom Control Editor plugin](https://plugins.octoprint.org/plugins/customControl/)?

If you can mock it up, then perhaps someone can take a look at what it would take to code this. Otherwise, we're sort of flying in the dark here, I'd guess.

Hywelmartin, 2019-03-31:
I used the Custome control editor and made my buttons..
I dont really know how this would help.. but here they are..
[image]

I found some settings that might be the place to alter.. if I remove M0, M1, M25,and add my Paus macro I believe that would sort out the Pause.. but resume...??
[image]

OutsourcedGuru, 2019-04-10:
I think what would be useful for someone considering to perform the work would be more like:

pause: `PAUSE`

...or whatever commands are necessary in that situation. Since few/none of us have Klipper, we'd be flying blind here.

cniweb, 2019-04-10:
I would wish for a break that allows you to turn off the printer and after turning on a resume of the print job at the position where the break was triggered.

cniweb, 2019-04-10:
@foosel has already designed a similar plugin:
https://github.com/foosel/OctoPrint-PrintRecoveryPoc

closed 2019-11-28 by Hywelmartin