OctoPrint/OctoPrint #3044: Cant use octoprint with Firefox anymore

rmx-it, 2019-02-11:
#### What were you doing?

After one of the last updates of octoprint I cant use octoprint with Firefox anymore.
Before the octoprint update it did work. OS is FreeBSD but shouldn't matter here.

#### What did you expect to happen?

I would expect to see a funtional octoprint UI.

#### What happened instead?

I see an quite basis screen like the Javascript parts where missing.
No buttons work.
Chrome does work though.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.3.10 running

#### Operating System running OctoPrint

OctoPi 0.13.0

#### Printer model & used firmware incl. version

CoreXY with Repetier FIRMWARE_NAME:Repetier_0.92.9

#### Browser and version of browser, operating system running browser

FreeBSD either 11 or 12

#### Link to octoprint.log

https://pastebin.com/78kQfvrK

#### Link to contents of terminal tab or serial.log

https://pastebin.com/K0UrSucQ

#### Link to contents of Javascript console in the browser

https://pastebin.com/cLt7WxA9

#### Screenshot(s)/video(s) showing the problem:

See https://imgur.com/ZtVLmeb

I have read the FAQ.

foosel, 2019-02-11:
I see that you are using OctoPrint via a public URL. Besides this [not being recommended](), please check whether the issue persists if you connect *directly* to your instance. If that works fine, something related to that forward/proxy setup is at fault.

A screenshot of your debug console's network tab would also be interesting. From the looks of your JS console it certainly feels that huge parts of the site JS cannot be loaded or executed, leading to mandatory parts missing and hence stuff not loading:

```
Error: settingsViewModel is missing, can't run UI packed_core.js:15834:19
```

Also check if clearing your browser cache helps.

DanielJoyce, 2019-02-11:
Just fire up maintenance branch in latest FF on Linux, no problems.

rmx-it, 2019-02-12:
Hi,
no its not a public URL. I just use a internal name server for an internal domain.
All communication is done directly. Both firefox client and octoprint are in the same IP segment.

network tab after clearing and disabling cache: https://imgur.com/a/n4lxElf

foosel, 2019-02-13:
@rmx-it and there I hoped for some obvious errors in there. Is there any chance I could take a look at this instance myself, e.g. through some kind of tunnel or something? At this point I fear I'm out of ideas what could be going on here, and poking around on it, maybe debugging a bit through the UI initialization might help.

Jurnee58, 2019-03-06:
Has this issue been resolved yet? I'm having the exact same problem. I'm running my Octoprint instance on my Raspberry Pi 3 and it was fine up until about a week ago when I updated Octoprint. Now when I try to connect to it through Firefox on my Windows 10 machine I get the exact issue @rmx-it describes. It works in Chrome, but not Firefox.

foosel, 2019-03-06:
No, it hasn't been resolved yet because I cannot reproduce it, so far no one else but @rmx-it reported it and my question for some way to take a look at the affected instance myself in order to debug never got an answer either 🤷‍♀️

Ipirk88, 2019-03-14:
I have the same Problem - i have to clean everytime the Cash form FF :-(
To clean everytime the Cash is a fix but a very bothersome fix

foosel, 2019-04-05:
Anyone running into this to actually work with me to look further into it? Otherwise I fear I'll have to close it. I use FF as my main browser on both desktop and mobile and it's working flawlessly here.

nlohr1, 2019-04-06:
Hi, my Firefox is as same rejecting - not the login, but viewing the GUI completely. I am using it in private mode and having problems also on other (completely different) sites with it after logging in.
FF Does not show the complete GUI, I cant load files nor see the status. My FF-version: 66.0.2 (64-Bit).
I will try to upload here a Screen *after* logged in (!):
[image]
I can login with an older Browser: Waterfox, but doubt to be secure with it...
If I can help to find out working together with You, please instruct me what to do.

foosel, 2019-04-08:
@nlohr1 you could start with providing a fully filled out ticket template ;) And testing if the problem persists in safe mode. And if there is any kind of way for me to take a look at an affected instance in person that would also help.

In your screenshot it looks like the GUI initialization ran into some kind of error, so the JS console might help to determine what is going on.

nlohr1, 2019-04-08:
Thank You very much, Gina! Your assumption that it was a fault *in Firefox* helped me out:
after erasing FF completely and installing it new from download solved my above mentioned problem!
I did never expect that FF can become faulty! (perhaps awith an "undercover"-malware?) --- Anyhow - sorry for my assumption of Your good Octoprint-GUI! This is now working perfectly like doing before and login etc. comes all up as expected! Thank you!

rmx-it, 2019-04-08:
Hi,
unfortunately most remote-access Software is designed for Windows/Linux and not for FreeBSD.
But if you would like to trace the problem on my client I could arrange a direct VNC session.
I would have to prepare a access/forwarding rule on the Firewall.
Than we could arrange a telephone session per PM and exchange the connection parameters...

foosel, 2019-04-11:
@rmx-it VNC would work for me. Though I'd prefer something that isn't a telephone for coordination in between - I try to avoid giving out my phone number and I also don't have a headset for that ;) Chat would work for me there, or any one of the multitude of voice chat solutions out there (which are probably not available under FreeBSD either though, but maybe you have a different OS in the house somewhere?).

rmx-it, 2019-04-16:
Hi,
I could create a Webex for chatting. I have XMPP too.

foosel, 2019-05-03:
Turns out it is more like 3 weeks that are kinda crazy... would mid to late next week work for you?

foosel, 2019-05-09:
@rmx-it ping :)

rmx-it, 2019-05-11:
Hi,
it seems that I have been as busy as you :-)
Next week I'm busy until friday too.
So maybe next friday or the week after the next week would be fine.

foosel, 2019-05-15:
@rmx-it I'd like to send you a whenisgood link which will make this way easier I think, but might make more sense to do that on a private channel (I don't know about you but *I* don't want have everyone and their mother a clear insight into my availability hours ;)). Can you shoot me an email at gina -at- octoprint.org?

ungutknut, 2019-05-30:
I'm experiencing a similar issue on Firefox 67 but it differs a bit from the described problem (so I'm not sure if It'd be better to open a new bug for it): The GUI does work after logging in, but it's drastically slower than usually and after some time running it takes ages (>xx seconds) for the GUI to load/respond.

When I start the browser and log in, everything's alright but after some time (xx minutes) the "Web Content" process utilizes one CPU core for 100% and octoprint behaves extremely sluggish (with no increase in CPU utilization on the server).

It seems this started with 1.3.11 and I don't think it's an firefox-update that triggered this behaviour - although I'm not entirely sure if it also was that bad on FF <67. BTW this happens on two different linux clients and with 2 different octoprint servers (both RPi3B+ and 1.3.11; one fresh install, the other one updated from the pre-previous release).

I've also tried safe-mode (both, octoprint and firefox) which didn't help.

Here's a debug-logfile; One occurence of sluggishness that I noticed was at around 19:30. I restarted the browser then which got rid off the problem temporarily.
[octoprint.log](https://github.com/foosel/OctoPrint/files/3238143/octoprint.log)

foosel, 2019-06-03:
@ungutknut that looks like a different issue. Could you please open a new ticket for that? The only thing that I see in this log which looks like something is the browser disconnecting and reconnecting around 19:33, but I guess that was you closing the browser and reopening it? The JS error console and the network tab from the browser dev tools could contain more hints.

foosel, 2019-06-03:
@rmx-it Did you send me an email? If so it never arrived... 🤔

rmx-it, 2019-06-10:
Hi,
yes I did send an email.
But in the meantime after updating to the latest octoprint, it seems to work again on Firefox/FreeBSD.
So many thanks for your support anyway!

closed 2019-06-10 by rmx-it

foosel, 2019-06-11:
Hm, odd... but I'll take it ;) Also apologies - I just found your mail in my spam folder. Of course 🙄