OctoPrint/OctoPrint #3046: [Request] System notifications

larsemil, 2019-02-12:
Hi and thanks for a awesome software. I love it.

One thing i would love is for the notifications that OctoPrint is creating to be system wide. 
You know, like Facebook does it. You have to allow notifications and after that you get them system wide.

Might be this? https://developer.mozilla.org/en-US/docs/Web/API/Notifications_API/Using_the_Notifications_API

Thanks again for a awesome project

dovecode, 2019-06-05:
What notifications do you think would warrant a system toast? If you look at the existing notification-related plugins, the focus there is on printing events (print started, cancelled, finished, etc). Those are the ones I think would be useful, too, but maybe you're talking about the announcements ("New plugin available", "New version of OctoPrint", etc)?

Fwiw, the former would be easy to do as a plugin. The latter I don't think there are event mixins for...

KTibow, 2022-01-17:
Trying to make a plugin for this, but I couldn't find any good tutorials on handling the JS side. Anyone have any tips for learning how OctoPrint's UI works?