OctoPrint/OctoPrint #3047: 'TypeError: 'bool' object is not callable' after login page

koenkooi, 2019-02-12:
#### What were you doing?

I'm switching from a few months old 'devel' to the current 'maintenance' branch. After filling in user/passwd and clicking login I get these errors:

With caches and cookies cleared:

```
Feb 12 12:14:01 blackbox octoprint[10085]: 2019-02-12 12:14:01,803 - octoprint - ERROR - Exception on / [GET]
Feb 12 12:14:01 blackbox octoprint[10085]: Traceback (most recent call last):
Feb 12 12:14:01 blackbox octoprint[10085]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
Feb 12 12:14:01 blackbox octoprint[10085]:     response = self.full_dispatch_request()
Feb 12 12:14:01 blackbox octoprint[10085]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
Feb 12 12:14:01 blackbox octoprint[10085]:     rv = self.handle_user_exception(e)
Feb 12 12:14:01 blackbox octoprint[10085]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
Feb 12 12:14:01 blackbox octoprint[10085]:     reraise(exc_type, exc_value, tb)
Feb 12 12:14:01 blackbox octoprint[10085]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
Feb 12 12:14:01 blackbox octoprint[10085]:     rv = self.dispatch_request()
Feb 12 12:14:01 blackbox octoprint[10085]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
Feb 12 12:14:01 blackbox octoprint[10085]:     return self.view_functions[rule.endpoint](**req.view_args)
Feb 12 12:14:01 blackbox octoprint[10085]:   File "/usr/lib/python2.7/site-packages/octoprint/server/views.py", line 404, in index
Feb 12 12:14:01 blackbox octoprint[10085]:     if plugin.will_handle_ui(request):
Feb 12 12:14:01 blackbox octoprint[10085]:   File "/usr/lib/python2.7/site-packages/octoprint/plugins/forcelogin/__init__.py", line 52, in will_handle_ui
Feb 12 12:14:01 blackbox octoprint[10085]:     result = passive_login()
Feb 12 12:14:01 blackbox octoprint[10085]:   File "/usr/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 571, in passive_login
Feb 12 12:14:01 blackbox octoprint[10085]:     if user is not None and not user.is_anonymous() and user.is_active():
Feb 12 12:14:01 blackbox octoprint[10085]: TypeError: 'bool' object is not callable
Feb 12 12:14:01 blackbox octoprint[10085]: 2019-02-12 12:14:01,852 - tornado.access - ERROR - 500 GET / (172.20.0.85) 11132.33ms
```

In a tab that was logged into the previous 'devel' instance:

```
Feb 12 12:11:36 blackbox octoprint[10036]: Traceback (most recent call last):
Feb 12 12:11:36 blackbox octoprint[10036]:   File "/usr/lib/python2.7/site-packages/octoprint/server/util/sockjs.py", line 270, in _emit
Feb 12 12:11:36 blackbox octoprint[10036]:     proceed = proceed and hook(self, self._user, type, payload)
Feb 12 12:11:36 blackbox octoprint[10036]:   File "/usr/lib/python2.7/site-packages/octoprint/plugins/forcelogin/__init__.py", line 177, in socket_emit_validator
Feb 12 12:11:36 blackbox octoprint[10036]:     if user is not None and not user.is_anonymous() and user.is_active():
Feb 12 12:11:36 blackbox octoprint[10036]: TypeError: 'bool' object is not callable
```

#### What did you expect to happen?

Login succeeds, main page loads

#### What happened instead?

Error 500 in browser, Traceback in logs

#### Did the same happen when running OctoPrint in safe mode?

Same behavour in safe mode
```
Feb 12 12:17:26 blackbox octoprint[10145]: 2019-02-12 12:17:26,263 - octoprint.startup - INFO - ******************************************************************************
Feb 12 12:17:26 blackbox octoprint[10145]: 2019-02-12 12:17:26,278 - octoprint.startup - INFO - Starting OctoPrint 1.3.11.dev197+gc6b4eee.dirty
Feb 12 12:17:26 blackbox octoprint[10145]: 2019-02-12 12:17:26,279 - octoprint.startup - INFO - Starting in SAFE MODE. Third party plugins will be disabled!
Feb 12 12:17:26 blackbox octoprint[10145]: 2019-02-12 12:17:26,281 - octoprint.startup - INFO - Reason for safe mode: command line flag
Feb 12 12:17:26 blackbox octoprint[10145]: 2019-02-12 12:17:26,282 - octoprint.startup - INFO - ******************************************************************************
```

#### Version of OctoPrint

maintenance branch, c6b4eee437ce2e95f39b61ce859fc8097d17ed47, "1.3.11.dev197+gc6b4eee"

#### Operating System running OctoPrint

Linux 4.18.5 on a beaglebone black

#### Printer model & used firmware incl. version

Fabricator mini with Marlin bugfix-2.0.x

#### Browser and version of browser, operating system running browser

Safari on OSX
Chrome on OSX

#### Link to octoprint.log

https://gist.github.com/koenkooi/a112559055571234cd0474d700454dbd

I have read the FAQ.

foosel, 2019-02-12:
Have you run `python setup.py clean` and `pip install .` after switching?

koenkooi, 2019-02-12:
No, octoprint isn't installed using pip, but using actual package management from the distro. (Which I want to change, but venvs don't seem to like being built in one location and deployed in another, but that's a problem for another day.) 

Your comment did prompt me to rule out issues with with stale files, so I uninstalled octoprint and everything python related and reinstalled everything. The only things that were kept were config.yaml and the octoprint homedir.

While that was running I managed to get ipv6 working.

Behold:

```
Feb 12 17:05:22 blackbox octoprint[10463]: 2019-02-12 17:05:22,330 - octoprint.server.util.sockjs - INFO - New connection from client: 2a01:90a0:1ff:f100:<redacted>
Feb 12 17:05:22 blackbox octoprint[10463]: 2019-02-12 17:05:22,904 - octoprint.server.util.flask - INFO - Passively logging in user nynke from 2a01:90a0:1ff:f100:<redacted>
```

While I would like to pretend the ipv6 change fixed this, I strongly suspect I somehow managed to create a faulty installation.

Closing as PEBKAC :)

closed 2019-02-12 by koenkooi