OctoPrint/OctoPrint #3049: Software update fails on Debian testing/unstable

rgriebl, 2019-02-13:
Tested with Octoprint 1.3.10

Right now, the software update will not work in Debian testing and unstable. The problem here is that Debian decided to name its python version `2.7.15+` (yes the '+' is part of the version string returned by `python --version`. Octoprint's version compare cannot cope with this and sees this as < 2.7.9 (which leads to a very confusing output of the software update component telling you why it doesn't want to update: `Python: 2.7.9 (you have: 2.7.15+)` ... took me some time to figure out what is actually going on)

There is a bug report for Debian's python2.7 package already here https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=914072
Sadly there is no feedback from the Debian maintainers on why they named it this way and if this might be changed again in the future.

foosel, 2019-02-14:
Ew. PEP440 non compliant version numbers. I'll take a look if I can work around this somehow without breaking version detection of PEP440 *compliant* numbers.

rgriebl, 2019-02-17:
I really needed to get my MQTT plugin updated, so I dug a bit deeper and fixed it :)

foosel, 2019-02-18:
Fixed by your commit, merged to `maintenance`, soon `devel`, will be released with 1.3.11. Thanks! 👍

draeath, 2019-04-18:
FYI, the debian bug has been addressed in `python2.7 2.7.16-1`:

```
Date: Mon, 04 Mar 2019 16:29:09 +0100
Source: python2.7
Architecture: source
Version: 2.7.16-1
Distribution: unstable
Urgency: medium
Maintainer: Matthias Klose <doko@debian.org>
Changed-By: Matthias Klose <doko@debian.org>
Closes: 914072
Changes:
 python2.7 (2.7.16-1) unstable; urgency=medium
 .
   * Python 2.7.16 release.
     - Now has a version without a trailing '+'. Closes: #914072.
```

PR #3054 (2019-02-18): Fix python version detection on Debian testing/unstable
The current 2.7 Python binary on Debian returns "2.7.15+" as its version, which
is not PEP440 compliant. Tthere is no feedback at all from Debian maintainers
regarding the bug report as of now:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=914072

This patch makes sure to drop any trailing '+' character in a version string.

Fixes #3049

  * [ ] Your changes are not possible to do through a plugin and relevant
    to a large audience (ideally all users of OctoPrint)
  * [ ] If your changes are large or otherwise disruptive: You have
  [... cut]
commit c35708a4 (2019-02-18): Fix python version detection on Debian testing/unstable

The current 2.7 Python binary on Debian returns "2.7.15+" as its version, which
is not PEP440 compliant. Tthere is no feedback at all from Debian maintainers
regarding the bug report as of now:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=914072

This patch makes sure to drop any trailing '+' character in a version string.

Fixes #3049
closed 2019-05-14 by foosel