OctoPrint/OctoPrint #3057: deleting a gcode file is very slow (about 20sec)

tedder, 2019-02-21:
#### What were you doing?
Deleting a gcode file, any gcode file, takes approximately 20 seconds. In safe mode it takes approx 16 seconds.

##### standard mode
Here's the `octoprint.log`, with everything turned to `debug` and my standard plugins installed, it took about 20sec.
```
2019-02-21 15:53:36,534 - octoprint.events - DEBUG - Firing event: FileDeselected (Payload: None)
2019-02-21 15:53:36,535 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6f0c2f10>>
2019-02-21 15:53:36,550 - octoprint.plugin - DEBUG - Calling on_event on Octoslack
2019-02-21 15:53:36,574 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
2019-02-21 15:53:36,576 - octoprint.plugin - DEBUG - Calling on_event on PrusaMeshMap
2019-02-21 15:53:36,578 - octoprint.plugin - DEBUG - Calling on_event on announcements
2019-02-21 15:53:36,580 - octoprint.plugin - DEBUG - Calling on_event on displayprogress
2019-02-21 15:53:36,583 - octoprint.plugin - DEBUG - Calling on_event on mqtt
2019-02-21 15:53:36,595 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2019-02-21 15:53:36,599 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2019-02-21 15:53:36,601 - octoprint.plugin - DEBUG - Calling on_event on printhistory
2019-02-21 15:53:36,602 - octoprint.plugin - DEBUG - Calling on_event on smsnotifier
2019-02-21 15:53:36,604 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2019-02-21 15:53:36,606 - octoprint.plugin - DEBUG - Calling on_event on thespaghettidetective
2019-02-21 15:53:40,503 - octoprint.plugin - DEBUG - Calling on_event on tracking
2019-02-21 15:53:52,417 - octoprint.events - DEBUG - Firing event: FileRemoved (Payload: {'path': u'98_e3_white_OP-Z_Knob_Alter_0.15mm_PLA_MK3MMU2.gcode', 'storage': u'local', 'name': u'98_e3_white_OP-Z_Knob_Alter_0.15mm_PLA_MK3MMU2.gcode', 'type': ['machinecode', 'gcode']})
2019-02-21 15:53:52,421 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6f0c2f10>>
2019-02-21 15:53:52,429 - octoprint.plugin - DEBUG - Calling on_event on Octoslack
2019-02-21 15:53:52,439 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
2019-02-21 15:53:52,439 - octoprint.plugin - DEBUG - Calling on_event on PrusaMeshMap
2019-02-21 15:53:52,440 - octoprint.plugin - DEBUG - Calling on_event on announcements
2019-02-21 15:53:52,440 - octoprint.plugin - DEBUG - Calling on_event on displayprogress
2019-02-21 15:53:52,444 - octoprint.plugin - DEBUG - Calling on_event on mqtt
2019-02-21 15:53:52,453 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2019-02-21 15:53:52,456 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2019-02-21 15:53:52,457 - octoprint.plugin - DEBUG - Calling on_event on printhistory
2019-02-21 15:53:52,458 - octoprint.plugin - DEBUG - Calling on_event on smsnotifier
2019-02-21 15:53:52,458 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2019-02-21 15:53:52,461 - octoprint.plugin - DEBUG - Calling on_event on thespaghettidetective
2019-02-21 15:53:54,388 - octoprint.plugin - DEBUG - Calling on_event on tracking
2019-02-21 15:53:54,420 - octoprint.events - DEBUG - Firing event: UpdatedFiles (Payload: {'type': 'printables'})
2019-02-21 15:53:54,488 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6f0c2f10>>
2019-02-21 15:53:54,538 - octoprint.plugin - DEBUG - Calling on_event on Octoslack
2019-02-21 15:53:54,562 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
2019-02-21 15:53:54,563 - octoprint.plugin - DEBUG - Calling on_event on PrusaMeshMap
2019-02-21 15:53:54,563 - octoprint.plugin - DEBUG - Calling on_event on announcements
2019-02-21 15:53:54,565 - octoprint.plugin - DEBUG - Calling on_event on displayprogress
2019-02-21 15:53:54,623 - octoprint.plugin - DEBUG - Calling on_event on mqtt
2019-02-21 15:53:54,632 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2019-02-21 15:53:54,661 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2019-02-21 15:53:54,677 - octoprint.plugin - DEBUG - Calling on_event on printhistory
2019-02-21 15:53:54,689 - octoprint.plugin - DEBUG - Calling on_event on smsnotifier
2019-02-21 15:53:54,739 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2019-02-21 15:53:55,274 - octoprint.plugin - DEBUG - Calling on_event on thespaghettidetective
2019-02-21 15:53:56,622 - octoprint.plugin - DEBUG - Calling on_event on tracking
2019-02-21 15:53:56,622 - octoprint.events - DEBUG - Firing event: UpdatedFiles (Payload: {'type': 'gcode'})
2019-02-21 15:53:56,625 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6f0c2f10>>
2019-02-21 15:53:56,627 - octoprint.plugin - DEBUG - Calling on_event on Octoslack
2019-02-21 15:53:56,638 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
2019-02-21 15:53:56,641 - octoprint.plugin - DEBUG - Calling on_event on PrusaMeshMap
2019-02-21 15:53:56,643 - octoprint.plugin - DEBUG - Calling on_event on announcements
2019-02-21 15:53:56,646 - octoprint.plugin - DEBUG - Calling on_event on displayprogress
2019-02-21 15:53:56,648 - octoprint.plugin - DEBUG - Calling on_event on mqtt
2019-02-21 15:53:56,653 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2019-02-21 15:53:56,654 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2019-02-21 15:53:56,657 - octoprint.plugin - DEBUG - Calling on_event on printhistory
2019-02-21 15:53:56,659 - octoprint.plugin - DEBUG - Calling on_event on smsnotifier
2019-02-21 15:53:56,660 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2019-02-21 15:53:56,662 - octoprint.plugin - DEBUG - Calling on_event on thespaghettidetective
2019-02-21 15:53:57,103 - octoprint.plugin - DEBUG - Calling on_event on tracking
```

##### safe mode
Here's the log in safe mode. Again this took ~16sec, but most of that occurrs before the first message even shows up in the logs.
[... 40 lines cut ...]
prusa i3 mk3 mmu2

#### Browser and version of browser, operating system running browser
chrome on ubuntu

#### Link to octoprint.log
As above.

#### Link to contents of terminal tab or serial.log
n/a.

#### Link to contents of Javascript console in the browser
Nothing in the JS console, though the network request tab shows my final delete took 15sec and returned a 204.

#### Screenshot(s)/video(s) showing the problem:
n/a, but if you want I can do a 'screen vid' showing the delay after clicking.

#### other
I don't know if a delay between two "calling" log entries mean the slow event is the one with the gap or the one before the gap, if that makes sense. I mean, this:
```
100 calling a
101 calling b
120 calling c
```
In that case, is it 'b' or 'c' that is slow?

foosel, 2019-02-21:
This is just a guess, but the issue is probably that when you delete a file it also removes the metadata for the file from `.metadata.json`. If you have a ton of other files in the same folder and that file is somewhat large due to that, serializing it back to disk after removing the offending entry takes a while.

It might be possible to postpone that cleanup and just mark the file as stale or something, then have a regular or just async cleanup job take care of that.

But it would be interesting to know first if this is really the reason, so if on your same setup you could try the same in a freshly created subfolder and see how long it takes there, that would be great.

tedder, 2019-02-21:
There were perhaps 6-8 files in there when I started timing the deletes, when I was watching the JS/network console it was the last file- so it was empty when completed. But.. I went and opened the file, it has 320k lines and is 9mb. Obviously it isn't removing the entries, so it's sort of [a Schlemiel problem](https://www.joelonsoftware.com/2001/12/11/back-to-basics/).

Analyzing the json, there are 230 files in there:
```
$ cat .octoprint/uploads/.metadata.json | jq "keys[]" | wc -l
230
```
And the vast majority of the file- 310k of 320k lines- is the "progress" array, which I assume is being filled in by PrintTimeGenius (right, @eyal0 ?).
```
$ cat .octoprint/uploads/.metadata.json | jq "values[]|.analysis.progress" | wc -l
309867
```

That makes me think the cleanup isn't taking place. Right? Once this print is done I'll any files, empy the metadata.json, then fill it with ~10 files and do the deletes. If that is still slow, I'll turn off 'genius' and try it again.

FWIW I don't use subfolders, I try not to keep old gcode on the pi.

tedder, 2019-02-21:
Hmm, a quick test says it is more than the json load that is slow, this takes less than 1sec to open and json-parse that file on the same Pi instance:
```
m = json.load(open(".octoprint/uploads/.metadata.json"))
print (m.keys())
```

tedder, 2019-02-21:
loading and then calling `copy.deepcopy` takes over 3sec.

I'm attaching a zipped copy of the 320k-line json here so you can test further.

[m.zip](https://github.com/foosel/OctoPrint/files/2890340/m.zip)

eyal0, 2019-02-21:
> is being filled in by PrintTimeGenius (right, @eyal0 ?).

Right

foosel, 2019-02-22:
> That makes me think the cleanup isn't taking place. Right?

I'd tend to agree, so this just escalated into a potential bug pending further analysis on Monday, when I'm back in a maintenance/debugging time slot ;)

Regardless of the cleanup taking place or not, the `deepcopy` situation needs a closer look too.

foosel, 2019-02-26:
So, first few results...

When I delete a file, it's successfully cleaning up the associated metadata over here. So that does appear to work in general.

Looking into what's making the deletion take so long, it's a combination of both *dumping* this amount of metadata and also - as already mentioned - the deepcopy made for the cache. The actual `dump` takes way longer than the `deepcopy` on my test though, 8s vs 1s on my desktop machine.

I'll look if it's possible to decouple the metadata cleanup from the actual deletion.

tedder, 2019-02-26:
Do you have any ideas where I can look for why it isn't being properly pruned?

foosel, 2019-02-26:
The only idea I have right now would be deleting those files outside of OctoPrint's control - the metadata isn't matched up on every start with what's on disk or anything like that, if you delete stuff outside of OctoPrint OctoPrint will never know about that and hence not delete the metadata entry. But that's about the only scenario I can currently come up with how stuff could get left over, apart from a server crash while metadata is not yet written back.

tedder, 2019-02-26:
Both of those are likely. I'll wipe the files and truncate the json and see what happens.

foosel, 2019-03-05:
I've decided against merging b300a45  into `maintenance` for now since it is still causing me massive issues in my tests. However, I've cherry picked 15d3b2c as 7d0bd6a into `maintenance`, which should at least get rid of stale entries in `.metadata.json` (which also appears to be a problem here).

I want to get the uncoupling stuff merged, but I'm not sure I'll manage to do so for 1.3.11 since I also have some other stuff on my plate for that and as mentioned in the latest OctoPrint On Air have reduced maintenance work to two days per week again as development work was getting shortchanged. And those two days were now eaten by trying to get the uncoupling to work reliably two weeks in a row ;)

tedder, 2019-03-05:
I'm 6wks behind on my vids, so I haven't seen it yet  :) But I did figure out a couple of things about this:
- truncating the JSON fixed it, of course.
- it's my fault it happened; I had so many files I just cleaned up all the older ones with `find`.
- obviously performance with large lists would still be a problem
- maybe a batch/uncoupled/async is the only way that deleted metadata needs to be pruned? I mean, there would be no need to do it in two places.

tedder, 2019-03-05:
dirty flag in b300a456fbaafeaad3978e7064533f1032a52806 FTW. It does seem like code smell to have so many deepcopies though.

foosel, 2019-03-06:
So I had an epiphany this morning under the shower 🚿 and switched from full deep copies to a combination of shallow copies of everything and deep copies only for the adjusted entries. Pushed above. Serialization to disk of large objects still is a tad slow, but it's way way better now. And it doesn't have this horrible added complexity of fully decoupling things that makes me want to bite my desk. 

Cleanup of stale entries is also still in. Merged the above into `maintenance` and will push it out with 1.3.11. Tests welcome :)

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel