OctoPrint/OctoPrint #3060: Octoprint + GPX, cancel print hangs in waiting for OK

zuhra031, 2019-02-21:
#### What were you doing?

1. press print
2. wait for print to start
3. press cancel in web gui

#### Version of OctoPrint

1.3.10

#### Operating System running OctoPrint

OctoPi 0.15.1 running on rpi 3b+ with new microsd

#### Printer model & used firmware incl. version

CTC Dual/3D a.k.a. Replicator 1

#### Browser and version of browser, operating system running browser

OS X 10.11, latest FF

It looks like a same problem as [#1139](https://github.com/foosel/OctoPrint/issues/1139) but in newer octoprint version.

foosel, 2019-02-25:
If you run into any kind of connection or communication issues with printers requiring the GPX plugin to work with OctoPrint, please open a ticket in the plugin's repository instead of here, or get in touch on the community forums. 

The GPX plugin completely replaces the underlying serial connection with its own wrapper and I have absolutely no insight into or debugging possibilities for that. 

If there is an actual issue on OctoPrint's end, I'm sure @markwal will tell me after initial analysis. But as things stand, I cannot do anything about GPX specific issues.

closed 2019-02-25 by foosel

WolfgangFahl, 2019-02-25:
Do I need to open another BR since I didn't use the GPX plugin?

markwal, 2019-02-25:
@WolfgangFahl yes, because "Same here" was incorrect.  Your issue is likely precisely indicated by the log line you pasted. It looks like an issue with the tracking plugin or with another plugin you have installed that generated a poorly formed event payload.  You'll want to provide the full form of information to make diagnosis precise.