OctoPrint/OctoPrint #3062: OctoPrint interface does not load on Chrome on Android

alexbaucom17, 2019-02-25:
#### What were you doing?

1. Enter the IP of OctoPrint on my phone web browser
2. The web interface partially loads and is non-responsive (buttons don't work, can't start prints, move to different tabs, etc.)

#### What did you expect to happen?
I expected the interface to work normally. The interface works correctly on my desktop computer still

#### What happened instead?
The interface was frozen and not responsive. Even while the interface is frozen on my phone it is functional on the desktop

#### Did the same happen when running OctoPrint in safe mode?
Yes. The interface remained responsive on my desktop during safe mode as well.

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
OctoPi 0.14.0

#### Printer model & used firmware incl. version
Prusa MK3 
FW: 3.5.1-1778

#### Browser and version of browser, operating system running browser
Phone with non-responsive interface: Google Pixel running Android 9 build PQ2A.190205.003 and Chrome 72.0.3626.105
Desktop with responsive interface: Window 10 Build 17134.345 with Chrome 72.0.3626.109
Tested on additional phone that had a responsive interface: Google Pixel 2 XL Android 9 build PQ2A.190205.002 and Chrome 72.0.3626.105

#### Link to octoprint.log
https://pastebin.com/cC9cbJQ6

#### Link to contents of terminal tab or serial.log
No data - happens even after rebooting OctoPrint without a connection to the printer so there is no terminal info

#### Link to contents of Javascript console in the browser
Output from desktop with responsive interface: https://pastebin.com/yc2cXYBp
Can't find any way to easily get the javascript logs from mobile without needing a bunch of extra tools. Let me know if this is critical and I can try to get the logs

#### Screenshot(s)/video(s) showing the problem:
Partially loaded and frozen interface from my mobile
https://drive.google.com/file/d/1_-yhYaYdrEDcONm62JH1bL44EuEnA0Dg/view?usp=sharing
Fully loaded and functional interface from my desktop
https://drive.google.com/file/d/1biQi1UnY-FShjeQPVle76gc7206aIQ6E/view?usp=sharing

I have read the FAQ.

### Some other notes
The interface definitely used to work on my phone. It stopped working in the last month or so sometime (I don't recall when exactly). I seem to remember updating to the latest version of OctoPrint sometime somewhat recently as well, so it is possible that the phone stopped working after the update, but I don't know that for sure. I would be happy to try downgrading to an older release to test this theory out if that would be helpful.

foosel, 2019-02-26:
> Phone with non-responsive interface: Google Pixel running Android 9 build PQ2A.190205.003 and Chrome 72.0.3626.105

I happen to own a Pixel myself, with those exact versions (just checked). I cannot reproduce this issue, stuff loads and initializes just fine here. Considering that things work fine with another Android device and the same browser on your end as well, it sounds hard to reproduce and limited to one browser?

I fear without something like the JS error console it will be next to impossible to even figure out what's up here, so if you could look into getting those, that might help.

Also your unresponsive instances looks similar to the symptoms [described in this FAQ entry](https://community.octoprint.org/t/the-web-interface-seems-stuck-not-responsive-broken/5675?u=foosel) - you should try the steps listed therein and see if it fixes things.

alexbaucom17, 2019-02-27:
Sounds good. I will try the steps in the FAQ and see about extracting the JS logs from my phone. Might be a couple days till I get around to it though.

hudsonlighting, 2019-03-01:
I have this same issue. I've identified that it only happens when I am on a different access point on my network, despite my laptop also being on a different access point which still works. Sometimes, I find that I cannot even access the IP address the octopi is on. I believe it isn't an Octopi issue though - I'm running 3 Octopi's and they all do exactly the same thing so I think it's some sort of WIFI issue. Do you think you are possibly on a different access point to your Octopi when this happens to you too?

alexbaucom17, 2019-03-03:
Alright, I was able to get the console logs by following [this guide](https://medium.com/anantha-krishnan-k-g/how-to-debug-your-website-in-android-chrome-29ee031b24e3) which didn't require installing a bunch of android development tools.

I tried loading the page two different times, once via the dev tools and once via the mobile device. The results were slightly different so I've included both logs, though I'm guessing the second one is the interesting one since it seems to have fully loaded in the dev tools system and does show some errors.

Log when loading web page directly through chrome dev tools (the screen never actually loaded on the dev tools page): https://pastebin.com/Nf6nipjn
Log when loading web page via mobile device and inspecting through chrome dev tools (screen loaded on the dev tools page): https://pastebin.com/Nn9T1JpJ

A couple other things to note - when loading the page via mobile, the USB debugging would lose connection right as the page tried to load - only after a few moments could I reconnect and get the logs. Not sure if that has any significance or not. Additionally, after sitting on the frozen page for ~5 minutes while I was typing this up, chrome crashed and gave me the Aw, Snap page.

As for the other suggestions, I read through the FAQ and tried the various steps like rebooting in safe mode and checking for authentication issues, but to no avail. Also, I looked into my network configuration and noticed that I did have my raspberry pi on a 2.4 GHz band on the router while my phone was connected to a 5GHz band, but when I connected my phone to the 2.4 GHz band, the problem persisted. I only have a single home router in my house so everything connects to the same access point.

Mavamaarten, 2019-03-12:
I cannot produce anything useful in terms of logs, but since recently I have the same issue. The webpage loads, but as soon as data comes through the webpage freezes and nothing can be tapped.

I'm using a OnePlus 6T on Android 9. The browser I'm using is Brave 1.0.84, which is based on chromium 72.0.3626.121. The same issue appears on Chrome, but on Firefox everything works as expected.

The issue has only appeared pretty recently. I'm on octoprint 1.3.10.

alexbaucom17, 2019-03-13:
Yes this sounds like exactly the issue I am seeing. The page loads, but then freezes right after loading and nothing is tappable. It only started happening recently for me as well (maybe in the last month or so). I'm considering trying to revert to a previous version of octoprint to see if the issue goes away. Just need to find the time to try that....

MichaelFlorer, 2019-03-14:
I have this same issue using Chrome on my laptop. I have noticed a temporary fix that works every time for me: Open an incognito tab, login to octoprint, then let it load. Go back to a normal browsing tab and load your Octoprint IP again. It should allow you to login as normal now.

alexbaucom17, 2019-03-16:
Alright I've got some more updates:
- I tried reverting my octoprint install to 1.3.9 but that did not fix the issue for me
- I tried the suggestion from @MichaelFlorer on my phone, but even attempting to visit OcotPrint in an incognito tab didn't work since the interface froze
- Since @Mavamaarten noted that they were successful with a non-chromium based browser, I tried installing Firefox on my phone and visiting the server there. This worked and the interface loaded as expected and was responsive.

This makes me think that it wasn't an update to OctoPrint that caused the issue, but maybe a recent update to chromium that has some incompatibility with OctoPrint.

So I think my workaround for now will be to use Firefox on my phone for my OctoPrint needs.

AmazeCPK, 2019-06-13:
I believe I'm having the same issue with chrome on Android. I tried the Firefox suggestion and it immediately worked. Thank you for the suggestion.

cebnetw, 2019-11-28:
I think I may have stumbled on a clue.  I was having the same issue using an android tab and chrome connecting via to octopi using the internal home network address but in another tab I could connect through the dmz address i have set up.  Chrome has a setting called lite mode.  I turned lite mode off and the problem disappeared on the home address tad but  appeared on the internet dmz address tab.   Turning lite mode back on did not cause the problem to come back to the internal address immediately.  Connetion only seems work on one address or the other at any particular time.

brandon3055, 2019-12-14:
May or may not be related but i have recently started experiencing a similar issue. Except in my case i am able to load octoprint on my phone and everything seems to work fine at first. But if i start using the controls to move around the hot end (e.g. for bed leveling) The page will just randomly die as if the web server on the pi has crashed. When this happens i am no longer able to access the page from my computer ether until i reboot the pi.

LazeMSS, 2021-03-31:
Anyone still having this issue? No posts since 2019...

LazeMSS, 2021-03-31:
Closing - if any of the original reporters are still having this issue then comment.

closed 2021-03-31 by LazeMSS