OctoPrint/OctoPrint #3063: Print cancel hangs in status "cancelling!

WolfgangFahl, 2019-02-26:
#### What were you doing?
1. Print
2. Abort

#### What did you expect to happen?
Abort of print. Regain control of printer.

#### What happened instead?
Message "Aborting" - in german stayed in Status area.
Control of printer did not return

#### Did the same happen when running OctoPrint in safe mode?
N.A. Didn't test

#### Version of OctoPrint

OctoPrint 1.3.10

#### Operating System running OctoPrint
Raspbian Stretch

#### Printer model & used firmware incl. version
Geeetech A10, Marlin firmware

#### Browser and version of browser, operating system running browser
Firefox 65.0.1 on Mac Os High Sierra 10.13.6

#### Link to octoprint.log
https://gist.github.com/WolfgangFahl/f0641af4666de7fdfc0586307ee15871

`2019-02-25 15:00:33,172 - octoprint.util.comm - INFO - Changing monitoring state from "Printing from SD" to "Finishing" 2019-02-25 15:00:33,206 - octoprint.plugin - ERROR - Error while calling plugin tracking Traceback (most recent call last): File "/home/wf/source/python/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.10-py2.7.egg/octoprint/plugin/__init__.py", line 230, in call_plugin result = getattr(plugin, method)(*args, **kwargs) File "/home/wf/source/python/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.10-py2.7.egg/octoprint/plugins/tracking/__init__.py", line 119, in on_event self._track_printjob_event(event, payload) File "/home/wf/source/python/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.10-py2.7.egg/octoprint/plugins/tracking/__init__.py", line 246, in _track_printjob_event elapsed = int(payload.get(b"time")) TypeError: int() argument must be a string or a number, not 'NoneType' 2019-02-25 15:01:02,647 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Pausing"`

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser
N.A.

#### Screenshot(s)/video(s) showing the problem:
N.A.

I have read the FAQ.

foosel, 2019-02-26:
I'm confused. You state that the print hangs in status `Cancelling` yet your log shows it going from `Printing from SD` to `Finishing` and then after trying to go from `Finishing` to `Pausing` (which indicates some other buttons pressed).

Also, a `serial.log` or terminal tab contents are missing which are needed to investigate any kind of communication issues. I'm not sure that the exception in your log snippet is actually the cause of whatever you are seeing because it gets caught and handled properly as far as I can see (though it should also be investigated). 

Please also *always* test in safe mode if an issue is reproducible there - because if it's not it's caused by one of your installed third party plugins which I have no insight into. Thanks.

WolfgangFahl, 2019-03-01:
Here is a serial log from this morning. I printed from SD and could not abort.
https://gist.github.com/WolfgangFahl/bf2f38fc3bd301a3e90e503a4fd767fc

foosel, 2019-03-01:
Your printer stopped responding to commands sent via serial altogether.

Notice how it tries to query an SD progress status from your printer via `M27` at `2019-03-01 10:31:15,485`, `2019-03-01 10:31:16,487` and `2019-03-01 10:31:17,490` without ever receiving an answer (as it still does for the `M27` sent at `2019-03-01 10:31:14,483` and responded to at `2019-03-01 10:31:14,707`):

```
2019-03-01 10:31:14,483 - Send: M27
2019-03-01 10:31:14,536 - Recv: Setting Print and Travel Acceleration: 5000.00
2019-03-01 10:31:14,707 - Recv: SD printing byte 7516/57947
2019-03-01 10:31:14,709 - Recv: ok
2019-03-01 10:31:14,711 - Recv: Setting Print and Travel Acceleration: 500.00
2019-03-01 10:31:15,485 - Send: M27
2019-03-01 10:31:16,434 - Recv:  T:203.47 /200.00 B:59.91 /60.00 @:0 B@:42
2019-03-01 10:31:16,487 - Send: M27
2019-03-01 10:31:17,490 - Send: M27
2019-03-01 10:31:18,434 - Recv:  T:202.15 /200.00 B:59.86 /60.00 @:0 B@:51
2019-03-01 10:31:19,027 - Recv: echo:enqueueing "M84 X Y Z E"
2019-03-01 10:31:20,432 - Recv:  T:201.08 /0.00 B:60.01 /0.00 @:0 B@:0
2019-03-01 10:31:22,432 - Recv:  T:198.98 /0.00 B:59.78 /0.00 @:0 B@:0
2019-03-01 10:31:24,433 - Recv:  T:197.61 /0.00 B:59.43 /0.00 @:0 B@:0
2019-03-01 10:31:26,432 - Recv:  T:196.08 /0.00 B:59.26 /0.00 @:0 B@:0
2019-03-01 10:31:28,412 - Recv: echo:enqueueing "G28"
2019-03-01 10:31:28,439 - Recv:  T:194.38 /0.00 B:58.78 /0.00 @:0 B@:0
2019-03-01 10:31:30,436 - Recv: echo:busy: processing
2019-03-01 10:31:30,447 - Recv:  T:192.98 /0.00 B:58.61 /0.00 @:0 B@:0
2019-03-01 10:31:32,436 - Recv: echo:busy: processing
2019-03-01 10:31:32,450 - Recv:  T:191.22 /0.00 B:58.42 /0.00 @:0 B@:0
2019-03-01 10:31:34,436 - Recv: echo:busy: processing
2019-03-01 10:31:34,442 - Recv:  T:189.35 /0.00 B:58.10 /0.00 @:0 B@:0
2019-03-01 10:31:36,348 - Recv: X:0.00 Y:0.00 Z:0.00 E:150.80 Count X:0 Y:0 Z:0
2019-03-01 10:31:36,438 - Recv:  T:188.07 /0.00 B:57.92 /0.00 @:0 B@:0
2019-03-01 10:31:38,437 - Recv:  T:185.83 /0.00 B:57.70 /0.00 @:0 B@:0
2019-03-01 10:31:40,438 - Recv:  T:184.42 /0.00 B:57.58 /0.00 @:0 B@:0
2019-03-01 10:31:42,438 - Recv:  T:182.81 /0.00 B:57.34 /0.00 @:0 B@:0
2019-03-01 10:31:44,437 - Recv:  T:180.82 /0.00 B:57.21 /0.00 @:0 B@:0
2019-03-01 10:31:46,437 - Recv:  T:179.27 /0.00 B:56.98 /0.00 @:0 B@:0
2019-03-01 10:31:47,947 - Changing monitoring state from "Printing from SD" to "Cancelling"
2019-03-01 10:31:48,438 - Recv:  T:177.75 /0.00 B:56.89 /0.00 @:0 B@:0
2019-03-01 10:31:50,437 - Recv:  T:176.35 /0.00 B:56.81 /0.00 @:0 B@:0
2019-03-01 10:31:52,437 - Recv:  T:174.84 /0.00 B:56.67 /0.00 @:0 B@:0
2019-03-01 10:31:54,437 - Recv: T:173.13 /0.00 B:56.41 /0.00 @:0 B@:0
```

The printer appears to briefly enter a blocked state (`busy:processing`), apparently while cooling down, but then stops sending even the `busy:processing` messages and only sends temperature reports back while continuing to cool down. In all that time OctoPrint never receives an `ok` or any other indication from the printer that it's ok to continue sending commands to it now.

Since in order to cancel OctoPrint needs to be able to send a cancel command to your printer which it never can here, it gets stuck in "Cancelling". That is factually correct - from OctoPrint's point of view, it's still waiting for your printer to respond again and telling it it's ok to send data over in order to proceed with the cancelling routine. Your printer, or rather its firmware, never does though.

It would be interesting to see what commands went to the printer at that point in your GCODE file, but sadly since you were printing from SD on not streaming from OctoPrint that will stay a bit of a mystery.

Long story short, this is in all likelihood either a firmware bug or a browning out printer. OctoPrint is behaving correctly here. Since there's nothing actionable for OctoPrint here I'm closing this, comments are still open though.

closed 2019-03-01 by foosel

WolfgangFahl, 2019-03-02:
Since I don't have another printer at this time I'd love to see a workaround that is better than fully restarting octoprint which I have to do now. I don't know whether I'll get a fix from the printer manufacturer and there might be other misbehaving devices out there ... At least octoprint should warn about the issue and  have an option to force back control to the GUI if the user deems this is the correct thing to do.

foosel, 2019-03-02:
There's *nothing* that OctoPrint can do or "return control" somehow here. Your printer stops responding on the serial interface. Since that is the only way OctoPrint can talk to it, it can't do anything. And it also cannot detect the risk of this happening before it happens so it can't warn against it either. You shouldn't have to restart however, simply disconnecting from and reconnecting to the printer shouldn't suffice. Be aware however that depending on your printer controller that will also restart the printer (most of them are *wired* to reset on start of a serial connection, yes that sucks, no it cannot be detected prior and warned against).

This issue is as if you were on the phone with someone and the other side simply stops talking to you. The only thing you can do in such a case is hang up 🤷

With regards to support from your printer's manufacturer, I suggest to look online for a community dedicated to your particular printer - usually there is at least one and there you can also usually find information on such problems, if there are solutions and links to alternative firmwares.