OctoPrint/OctoPrint #3067: Serial log not written even though enabled

nionio6915, 2019-03-01:
#### What were you doing?

Trying to connect to printer. All connection attempts will timeout. Terminal response stops as it displays the responses from the controller.  

#### What did you expect to happen?
Open serial log and begin troubleshooting. 

#### What happened instead?
Serial log is filled with statements to the effect that the serial logging is not enabled. 

#### Did the same happen when running OctoPrint in safe mode?
Yes. Serial log says serial logging not enabled. 

[image]

#### Version of OctoPrint

OctoPrint 1.3.10 
#### Operating System running OctoPrint
running on OctoPi 0.15.0

#### Printer model & used firmware incl. version

Eustahios V2, Marlin 2.0 bugfix

#### Browser and version of browser, operating system running browser
N/A

#### Link to octoprint.log
All logs in this zip file
[octoprint_logs.zip](https://github.com/foosel/OctoPrint/files/2917976/octoprint_logs.zip)

#### Link to contents of terminal tab or serial.log

All logs in this zip file
[octoprint_logs.zip](https://github.com/foosel/OctoPrint/files/2917976/octoprint_logs.zip)

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).
-->
All logs in this zip file
[octoprint_logs.zip](https://github.com/foosel/OctoPrint/files/2917976/octoprint_logs.zip)

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.
Yes. Many times.

foosel, 2019-03-01:
I just tried this on 1.3.10 and it works flawlessly and as expected over here:

```
2019-02-20 18:30:34,387 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
2019-02-28 16:24:48,713 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
2019-03-01 09:59:10,670 - Enabling serial logging
2019-03-01 09:59:13,945 - Connecting to: /dev/ttyACM0
2019-03-01 09:59:13,963 - Changing monitoring state from "Offline" to "Opening serial port"
2019-03-01 09:59:13,969 - Connected to: Serial<id=0x6bc82110, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2019-03-01 09:59:13,970 - Changing monitoring state from "Opening serial port" to "Connecting"
```

Just to make sure: I assume you did click "Save" after setting that checkbox?

nionio6915, 2019-03-01:
i will double check that i did.

nionio6915, 2019-03-01:
i checked save again and now it seems to be working. 

close case please

foosel, 2019-03-01:
Closed as requested

closed 2019-03-01 by foosel